prusa3d/PrusaSlicer #2091: [Feature request] - update stl/model from disk and keep the modifier meshes

neophyl, 2019-04-08:
Would it be possible to make the reload from disk feature keep the modifier data once they have been placed on to the plate.

I often find myself placing a model, positioning it/rotating and then adding support modifiers (enforcers and blockers etc).  I then do the slice and check the results look good.  Its at that point that the slice shows me that my model isn't really the best optimised for printing, like an overhang I hadn't quite removed enough of or similar.  At that point I make a really minor change to the original model and export the stl again.  

Unfortunately I then have to delete the old version, add the new version and redo all the modifiers on the newly updated stl file.  It would make things much quicker if I could right click on the stl and reload from disk without losing everything.

bubnikv, 2019-04-08:
it's in the planning, but not that simple to solve.

bubnikv, 2019-04-13:
Duplicate of https://github.com/prusa3d/Slic3r/issues/1890
Closing this one.

closed 2019-04-13 by bubnikv