prusa3d/PrusaSlicer #2093: Print tubes as one-layer

Bobo329, 2019-04-08:
### Version
1.42.0-beta

### Operating system type + version
Windows

### 3D printer brand / version + firmware version (if known)
Custom

Is there a way to slice tubes a different diameters and single layers and not multiple ones?

For example: I have two tubes one of diameter 0.5mm and one 1.0mm. The 0.5mm one is printed as a single pass while the 1.0mm takes three layers. I want the 1.0mm tube to be done in one layer at a slower speed so the diameter is correct.

sergeysi779, 2019-04-09:
I think it can be done by adjustting "External perimeters" extrusion width in "Print Settings" > Advanced.

Bobo329, 2019-04-12:
When I adjust this setting, while using the same extruder for both diameter tubes, the only thing that changes is the speed of the print. The larger diameter (1.0 mm) tube remains as three layers.

bubnikv, 2019-04-13:
Something like this?

[image]

Please try the attached 3MF and see the attributes assigned to each of the cone.

[cones-different-wall-thickness.zip](https://github.com/prusa3d/Slic3r/files/3075913/cones-different-wall-thickness.zip)

bubnikv, 2019-05-04:
did it help?

rtyr, 2020-04-17:
Answered. No response. Closing.

closed 2020-04-17 by rtyr