prusa3d/PrusaSlicer #2094: bad allocation

brentboyd, 2019-04-09:
### Version
_Version of Slic3r Prusa Edition used goes here_ 1.42.0-beta1

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_windows
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) I3 MK3 3.7
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ stock

### Behavior
* _Describe the problem_ went to slice and got Bad Allocation
* _Steps needed to reproduce the problem_ slice it agian
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_ yes

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-04-10:
Would you please provide your 3MF file? (go to file->save project)
How much RAM do you have? Did you run the 32bit or 64bit build?

brentboyd, 2019-04-10:
i am using the 32bit

bubnikv, 2019-04-10:
on a 32bit windows? How much RAM do you have on your system?

closed 2019-04-10 by brentboyd