prusa3d/PrusaSlicer #2096: chyba

michalmichalcz, 2019-04-10:
kg/k### Version
_Version of Slic3r Prusa Edition used goes here_

dobrý den používám slic3r prusa editon
a je tam chyba tam zadam 369 kg/korun
a ono mito spocita blbost 5793.84 opravte to prosim

espr14, 2019-04-11:
Která verze Slic3r? Jinak se tam zadává money/kg. Počítá to chybně s libovolným modelem nebo pouze jedním konkrétním?

michalmichalcz, 2019-04-12:
Váš slic3r pe 1.41.3+win64
S libovolným
-- 
děkuji s pozdravem
Dolíhal Michal
dolihal@email.cz

---------- Původní e-mail ----------

Komu: prusa3d/Slic3r <Slic3r@noreply.github.com>
Datum: 11. 4. 2019 22:36:18
"

(https://github.com/prusa3d/Slic3r/issues/2096#issuecomment-482301613), or 
mute the thread
"

espr14, 2019-04-12:
Verze 1.42.0 beta 1 tento problém nemá.

michalmichalcz, 2019-04-12:
Mě to počítá špatně

Datum: 12. 4. 2019 16:55:30
"

bubnikv, 2019-04-13:
Prosím vyexportujte 3MF (menu plater->export as 3MF) a přiložte sem.

bubnikv, 2019-05-04:
no response, closing

closed 2019-05-04 by bubnikv