prusa3d/PrusaSlicer #2097: Missing Sliced Info

hon4500, 2019-04-10:
### Version
1.41.3+
_
### Operating system type + version
Mac0S V10.14.3

### 3D printer brand / version + firmware version (if known)
_Prusa i3 Mk3S

### Behavior
Generally the "Sliced Info" window/view are not shown after finishing the Slicing.
It only shows after exporting G-code.
[image]

lukasmatena, 2019-04-10:
This issue has already been discussed, it is fixed in the recent alpha/beta releases of 1.42.
Closing.

hon4500, 2019-04-12:
Sorry but I can't see that this issue is fixed.
Just installed latest beta, and it still does not show "Sliced info" after slicing.
It shows after pressing Print or pressing Export...

bubnikv, 2019-04-13:
>Just installed latest beta

Which one? This is the current latest beta.
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1

bubnikv, 2019-05-04:
no response, closing

closed 2019-05-04 by bubnikv