prusa3d/PrusaSlicer #2098: 3D view does not respond the first time I try to rotate it

DrLex0, 2019-04-10:
### Version
1.42.0-beta1

### Operating system type + version
Mac OS X 10.14.4

### 3D printer brand / version + firmware version (if known)
Irrelevant

### Behavior
* After switching from 3D Editor View to Preview or vice versa, the 3D view does not respond to dragging unless it has been clicked at least once.
* _Steps needed to reproduce the problem_
  * Load any model and any settings.
  * Switch to Preview.
  * Drag the mouse across the 3D view.
  * Try dragging again.
  * Switch back to 3D editor view and try dragging the view.
* _Expected Results_
  * The 3D view moves every time I drag across it.
* _Actual Results_
  * After switching views, the 3D view only moves after it has been clicked.

Not a big problem but gets on the nerves quickly…

bubnikv, 2019-04-13:
Would you please confirm that it is fixed with this development version? Thanks.

[Slic3rPE-1.42.0-beta1+350-full-gb0c33a1-201904121629.dmg.zip](https://github.com/prusa3d/Slic3r/files/3075904/Slic3rPE-1.42.0-beta1%2B350-full-gb0c33a1-201904121629.dmg.zip)

DrLex0, 2019-04-13:
Yes, it is fixed. Thanks!

closed 2019-04-13 by DrLex0