prusa3d/PrusaSlicer #2099: Armband raft generated with holes

mranders, 2019-04-11:
### Version
Version 1.41.1+win64
### Operating system type + version
Windows 7 Professional SP1

### 3D printer brand / version + firmware version (if known)
Prusa MK2.5

### Behavior
* Scaled a model to 125%
* Use place plane on bed to align part with bed
* Use 4 layers of raft because part is organic
* Notice that raft has gap on layer 1-6 and then bridges the wrong way on layer 7
* _Expected Results_
* Correct raft
* _Actual Results_
* Hole-y raft
[image]

[image]
[image]

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

mranders, 2019-04-11:
3MF file of plater export
[ArmorRightBicep2_0.2mm_PLA_MK2.5.3mf.zip](https://github.com/prusa3d/Slic3r/files/3066915/ArmorRightBicep2_0.2mm_PLA_MK2.5.3mf.zip)

bubnikv, 2019-04-13:
Thanks for the report, it is reproducible.

bubnikv, 2021-02-22:
The incomplete raft has been fixed on the master, likely with fb7b9950501d01164955b7dd29f163005552234e

[image]

closed 2021-02-22 by bubnikv