prusa3d/PrusaSlicer #2101: First support layer does not stick to bed

espr14, 2019-04-11:
1.42.0 b1

There are 2 issues with first layer of support:
1. Sparse infill
2. Sharp corners

Space between lines decreases adhesion while sharp corners tend to bend during cooling. This picture is taken after print with very well cleaned bed. Such clean bed does not cause any bending of brims or printed part. Without extra cleaning the support makes printing impossible.

[image]

Solution:
1. provide setting for infill density for the first layer of support
2. pick (automatically) angle of infill so the number of corners is as low as possible
3. turn sharp corners to rounded ones

[image]

espr14, 2019-04-13:
#1165

espr14, 2019-04-21:
Hotfix:
1. Enable brim
2. Set support XY separation to 200% or more (this will get brim between support and object)

Solution proposal:
1. Add perimeters for support (enable/disable, width, height - for flex) so you can print object without brim but to print support with first (few) layers with perimeters (brim, sheath).
2. Allow changing infill type and density for first layer, intermediate layers and support interface.
3. Allow building support on objects's brim.

tkircher, 2019-05-23:
This is also related to #1156

bubnikv, 2021-02-24:
Implemented with fcb714c

The 1st layer support or raft will always have a perimeter drawn around it to improve support adhesion.

Also there will be new parameters raft_first_layer_density and raft_first_layer_expansion
to influence the 1st layer of raft or support.

closed 2021-02-24 by bubnikv