prusa3d/PrusaSlicer #2102: Layer height table has no effect

tkircher, 2019-04-12:
### Version
1.41.3

### Operating system type + version
MacOS

### Behavior
Setting a layer height override in the table has no effect at all. Often I want to set specific layer heights in specific areas, which can't be done with the smooth layer height editor.

lukasmatena, 2019-04-12:
The table has been removed from the 1.42 version - see the latest beta release. Quoting from the 1.42-alpha1 release log:

> Layer height table was dropped for now, though we plan to replace it with a more general feature to assign not only a layer height, but also other print parameters to layer spans. Assigning parameters to layer spans is implemented in Simplify3D through processes. While Slic3r supports the same feature through modifier meshes (slabs), having the option to define the parameters per layer spans table would be nice #495.

tkircher, 2019-04-12:
Ah, noted. Thanks for the response.

closed 2019-04-12 by tkircher

mkoistinen, 2019-05-11:
:+1: for getting the Static Layer Height Table back!