prusa3d/PrusaSlicer #2103: Prusa does random color switching

schneidp20, 2019-04-12:
### Version
I've tried a couple version with the same results
1.41.2 Final
1.42.0 Beta 1

### Operating system type + version
Windows 10 Pro 

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3, firmware: 3.7.0-2201

### Behavior
I'm not sure if I should submit this problem to you or Prusa.   I can't tell if it's the printer firmware or the generated Gcode.
The prusa stops printing to do a color switch.  I've seen this  happen off and on and up until now it's only been a nuisance.   Now I have a design that I basically can't print.  I've gotten only 4mm height in the print with 10 color switches!  I've tried reprinting, and tried several different SD cards.   I've also generated the gcode on 2 separate computer.   Same results.   I haven't tried it enough to notice if it happens at the same points every time.   If it would help, I could give you the layer where it happens.   It can also happen multiple times on the same layer. 

I'm basically stuck so I'm willing to provide anything to help resolve this problem.   Should I be contacting Prusa instead?

* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

lukasmatena, 2019-04-12:
>  I can't tell if it's the printer firmware or the generated Gcode.

It would be great if you attached a 3MF and your gcode (like the issue report template suggested), so that someone who can has data to work with.

Anyway, it could also be a false filament sensor alarm. Are you sure that it's not the case? (Try to disable the filament sensor, if it isn't already.)

schneidp20, 2019-04-12:
[Body58a.zip](https://github.com/prusa3d/Slic3r/files/3072991/Body58a.zip)

Sorry about no attachements.   

I'll try disabling the filament sensor but it'll be 10 hours before I try it.   BTW: I also tried reverting back to an earlier firmware (3.5.2) but same results.  

Thanks for the prompt response!

schneidp20, 2019-04-12:
It's the sensor.   Sorry to trouble you with the problem

closed 2019-04-12 by schneidp20