prusa3d/PrusaSlicer #2105: First layer infill density

espr14, 2019-04-12:
Could you add first layer infill density setting?

First layer is squished into bed to improve adhesion. But this causes lines to "fight" which causes artifacts. Lower density would decrease pressure so the lines are laid more evenly. Here is simulated 50% density. Lines are straight so it looks more uniformly. I suppose 60 - 80 % would be the best setting.

[image]

Here's the line fighting:

[image]

bubnikv, 2019-04-12:
I have had this discussion before. Some spacing is equivalent to setting up
your printer to print a bit higher. You can do that without Slic3r support.

espr14, 2019-04-12:
Setting first layer higher makes print look better but adhesion will decrease. I'll try higher bed temperature but I'm still limited by 110°C.

bubnikv, 2019-04-13:
I believe the extrusion is sticking so well with the print bed, because it
is pressed hard enough to it. This only happens, if the extrusion is
limited in directions, to which it can escape. That is, printing the first
layer line next to another line densely improves adhesion due to an
increased pressure. The downside is the elephant foot effect, which may be
corrected for in the software.

bubnikv, 2019-05-04:
The first layer height, first layer extrusion width and extrusion pressure are interconnected. There is no need for another parameter.

closed 2019-05-04 by bubnikv