prusa3d/PrusaSlicer #2106: "Detect thin walls" sounds strange to me

dzieva, 2019-04-12:
Here "Detect thin walls" is disabled
[image]

Here "Detect thin walls" is enabled
[image]

When "Detect thin walls" is enabled the walls get worse.

bubnikv, 2019-04-13:
please provide 3MFs to both of these exampls.

bubnikv, 2019-04-13:
I consider this to be duplicate of https://github.com/prusa3d/Slic3r/issues/2012

closed 2019-04-13 by bubnikv

dzieva, 2019-04-13:
> I consider this to be duplicate of #2012

I do not see resemblance to # 2012, or you or I'm crazy.