prusa3d/PrusaSlicer #2109: Retraction Speed not clear

FelixHauser, 2019-04-13:
### Version
1.42 alpha 7

### Operating system type + version
macOS 10.13.6

### 3D printer brand / version + firmware version (if known)
DIY Cartessian with Steel frame

### Behavior
Retraction speed values are not clear

Discussing with some 3d printing buddies of mine, I found out that the values for retraction speed are not fully clear. When I analyze the Gcode using gcode.ws I found two different values for retraction speed. One is the value I set up on the printer settings tab of Slic3r, and the other is always 80mm/s, no matter which speed I actually typed in. I tried with the same model with different slicers and, when I analyzed with gcode.ws it only showed the actually speed I set up.  
I don't know if it leads to some actual misbehavior of the printer or not.

[filament_settings.txt](https://github.com/prusa3d/Slic3r/files/3076215/filament_settings.txt)
[printer_settings.txt](https://github.com/prusa3d/Slic3r/files/3076216/printer_settings.txt)
[Separador.gcode.txt](https://github.com/prusa3d/Slic3r/files/3076217/Separador.gcode.txt)
[print_settings.txt](https://github.com/prusa3d/Slic3r/files/3076218/print_settings.txt)
[image]

bubnikv, 2019-04-13:
Some of the retraction is done during the wipe move. Do you have a wipe enabled?

FelixHauser, 2019-04-13:
Yes, I do have "wipe movement" enabled. I've just tried to disable it and see the analysis and, it just shows a retraction speed now... Ok, so that's been cleared up. My follow up question would be... why 85mm/s? or better formulated, what if my extruder motor can't handle that?

bubnikv, 2019-04-13:
> My follow up question would be... why 85mm/s? or better formulated, what if my extruder motor can't handle that?

The wipe speed is derived from the travel speed, set to some 80% or so of it.

> what if my extruder motor can't handle that?

That's a good question, slic3r does not ensure that. You have to cover that in your firmware.

FelixHauser, 2019-04-13:
That's quite weird if you ask me... On the section "machine limits" I have set the max extruder feedrate at 60mm/s, which is the same as it's set on the firmware. It seems the Slic3r chooses to not make use of it. Also, I've tried with different machines, with different limits, and this "second" speed has been always 85mm/s. I've the impression it's a hard-cooded value.

Anyway, now that I know it's related to the "wipe" feature, I can take into account.

Thanks!

bubnikv, 2019-04-13:
Everything is fine actually as long as you are using Marlin firmware and
you set the limits in the Machine limits table. In that case the Marlin
firmware clamps the extruder speed for you.

Sebastianv650, 2019-04-14:
@FelixHauser also keep in mind that gcode.ws is "lying" to you in the wipe cases. While the 35mm/s is the real retract / prime speed and equal to the extruder speed, the 80mm/s isn't. If you have a look into your gcode, you will see that the 80mm/s is the feed rate of the wipe move itself.
As this is a 3-dimensional move containing X, Y and E axis (or even 4 if you have bed leveling active), this speed of 80mm/s is the "total" speed of this move. Here is a small 2D example to make it clear, it works the same for 3 and 4 dimensions aka axis:

Say we have a 5mm move in X and 2mm in Y. As Phytagoras told us, the resulting travel move has a length of 5.4mm. This 5.4mm will be driven at the feedrate set by the G1 F command, for example 80mm/s. As a result, X will move at a speed of 74.3mm/s and Y at 29.7mm/s.

This means, you E axis speed isn't shown in this analyzer for wipe moves. But it's most likely way below 80mm/s. If you want to know it exactly, you can calculate it on your own now.

FelixHauser, 2019-04-15:
Thanks for your info @Sebastianv650 . I will take a screenshot of your comment and show it to my students next time they ask "why do we need math"...

closed 2019-04-15 by FelixHauser

guestisp, 2019-05-03:
> Say we have a 5mm move in X and 2mm in Y. As Phytagoras told us, the resulting travel move has a length of 5.4mm. This 5.4mm will be driven at the feedrate set by the G1 F command, for example 80mm/s. As a result, X will move at a speed of 74.3mm/s and Y at 29.7mm/s.

where did you get 74.3 and 28.7 based on your example ?

Sebastianv650, 2019-05-04:
By the magic of math, using the "sentence of phytagoras" (warning, free translation into english, might be wrong for native speakers) :-D
80=SQRT(74.3²+29.7²)