prusa3d/PrusaSlicer #2111: Seam position rear not being honored in 1.42 beta

x0rtrunks, 2019-04-13:
### Version
1.42 beta

### Operating system type + version
OSX Mojave

### 3D printer brand / version + firmware version (if known)
DuetWifi and RepRapFirmware on custom corexy, using firmware retraction.

### Behavior
* _Describe the problem_
 It's not honoring rear seam position on all layers, and instead placing the line start of the external perimeter a long travel distance away from the start of the inner perimeter.

With seam position set to rear, only the inner perimeter of some layers is positioned to the rear, the outer perimeter is positioned on an internal sharp corner or the front of the model causing ugly seam under runs and banding. 

It seems to only happen on layers that have small sections of solid fill near the front due to an overhang. It seems to prioritize being close to the solid fill rather than honoring seam position. This is leading to unnecessarily long travel moves right before starting the external perimeter. 

Adding extra prime on unretract can help seal up the seam, but since it's not doing this on ever layer it just causes blobs on the normal layers.

* _Steps needed to reproduce the problem_
Choose a model with solid fill near the front of the model and choose rear seam position. 

* _Expected Results_
I expect the seam position selection to be honored.

* _Actual Results_
Seam position is being modified by path optimization on a per layer basis.

  * _Screenshots from __*Slic3r*__ preview are preferred_
Screen shot showing unretracts on external perimeter.
[image]

Photo of print using nearest seam position
[image]

Photo of print using rear seam position
[image]

Photo of rear of model showing seam at rear on only some layers
[image]

#### Project File (.3MF) where problem occurs

Here's the .3MF 
[ZombieHunterHead_v4_90k.3mf.zip](https://github.com/prusa3d/Slic3r/files/3076645/ZombieHunterHead_v4_90k.3mf.zip)

guestisp, 2019-04-13:
OT: do you know where these huge layer lines came from ? i have to same exact issue and only with Slic3r.

x0rtrunks, 2019-04-13:
> OT: do you know where these huge layer lines came from ? i have to same exact issue and only with Slic3r.

I'm not sure what you mean.

guestisp, 2019-04-14:
Look at the posted pics, there are layer lines bigger than others, like inconsistant extrusion

x0rtrunks, 2019-04-14:
The inconsistent layers are a combination of the underruns from starting the line after a long travel and layer shifts from the nozzle contacting the curled up overhangs.

guestisp, 2019-04-14:
Do you know how to solve ?

x0rtrunks, 2019-04-14:
Well part of it is to use rear seam position so that all layers start in the same location, and part of it is to improve your overhang performance. But the whole point of this issue report is that rear seam position isn't behaving as it should.

x0rtrunks, 2019-04-18:
[image]

So here's the model printed in Cura with similar settings. Note how it obeys the Z seam positioning obeys the user defined position (rear) and no lines start on the front of the model.

Retraction settings are exactly the same using firmware retraction on the Duet.

guestisp, 2019-04-19:
I did, right now, a test print with seam set to rear and the seam was properly on the rear.

x0rtrunks, 2019-04-19:
Right, but was it the same model? I explained why I think it's not placing the seam at the rear on some layers that have some solid fill near the front of the model. I believe the path optimization is overiding the seam position. This won't happen on every model just as it doesn't happen on every layer of this model.

Try the 3mf I included and see if it does the same. 

Thank you for trying to reproduce.

I've also since tested with 1.42 beta 2 and 1.41 and the behaviour is the same.

guestisp, 2019-04-19:
No, it wasn't the same modem. I'm fighting against another issue seam-related and i've tried the rear positioning by change.

bubnikv, 2019-04-23:
the rear seam placement hint is weighted against other constraints like
heading the seam inside corners or not placing the seam over an overhang.

x0rtrunks, 2019-04-23:
Ok but does that explain why it would place the inner perimeter start seam at the rear and then travel all the way to the other side of the model to start the outer perimeter? It makes no sense, and isn't consistent or explained to behave like that anywhere, which would say it's a bug. The Cura print that forces the seam to the rear turned out much better, so if the rear seam is being weighed against other things it's clear it's not being given enough weight on it's own.

closed 2026-09-29 by prusaslicer-bot[bot]