prusa3d/PrusaSlicer #2112: Lift Z only when crossing perimeter

guestisp, 2019-04-13:
Currently, lift Z is bound to retractions. If we enable lift Z, at each retraction, Z is lifted.
As we have retract on layer changed enabled, on each layer change Z is lifted.

A new option to only lift Z when crossing perimeters would be useful

Timocop, 2019-04-15:
Its already there. "Print Settings > Infill > Only retract when crossing perimeters" Its enabled on default.

guestisp, 2019-04-15:
That's the issue, zhop is bond ti the retraction settings
We need to retract always, not only on perimeters but zhop during the infill is not useful

The same with layer change . If you have retract on layer change, you also have a zhop that could be useless

k3nnethl, 2020-05-12:
How tf is this a "very unusual idea"? This is espsecially useful when printing multiple parts in combinaton of avoid crossing parimeters. So when it jumps between parts it'll reduce/eliminate stringing.

bubnikv, 2020-12-11:
And there is a request for just
https://github.com/prusa3d/PrusaSlicer/issues/1854

bubnikv, 2020-12-11:
Lift Z is there mostly to avoid kicking objects off the print bed. If you are afraid of kicking objects off your print bed, then you should most likely Z-hop inside the island as well.

SimonSolar2C, 2021-09-26:
Actually I would love to have this option to avoid marks on the outer perimters of my model.  Knocking it off is an extreme case.
IE I would like to 'z-hop when crossing perimeters' - totally decoupled from other retraction settings

onolox, 2023-05-04:
Cura has a brillhant travel path engine, maybe look there to get some ideas.

closed 2026-09-29 by prusaslicer-bot[bot]