prusa3d/PrusaSlicer #2113: CMake can't find TBB on MacOS

tkircher, 2019-04-14:
### Version
version_1.42.0-beta1-352-g255a4e05d

### Operating system type + version
MacOS 10.12.6

### Behavior
CMake can't find TBB with the included FindTBB.cmake. A different version (copied from CGAL) worked correctly. My version of TBB was provided by macports, but it shouldn't matter.

bubnikv, 2019-04-14:
Did you compile with the SLIC3R_STATIC flag? Then the cmake expects TBB to be provided compiled statically as well, which I suppose the build from macports is not.

vojtechkral, 2019-04-15:
Yes, the `SLIC3R_STATIC` is by default set to `1` on Mac OS.

domesticatedviking, 2019-06-10:
Still having this issue.

domesticatedviking, 2019-06-10:
Found this which might be helpful: https://stackoverflow.com/questions/638278/how-to-statically-link-to-tbb

tkircher, 2019-06-11:
Just to update, setting the environment variable SLIC3R_STATIC to 0 solved the problem and everything builds successfully. Closing.

closed 2019-06-11 by tkircher

tkircher, 2019-06-11:
Incidentally, I just upgraded wxWidgets to 3.1.2, and SetWidth() is no longer a member of wxBitmap. So the instances of empty_bmp.SetWidth(0) used as a hack in GUI/Field.cpp and GUI/GUI_ObjectManipulation.cpp throw errors. I just deleted the calls to SetWidth() and everything seems to look fine. ;)

domesticatedviking, 2019-06-11:
> Incidentally, I just upgraded wxWidgets to 3.1.2, and SetWidth() is no longer a member of wxBitmap. So the instances of empty_bmp.SetWidth(0) used as a hack in GUI/Field.cpp and GUI/GUI_ObjectManipulation.cpp throw errors. I just deleted the calls to SetWidth() and everything seems to look fine. ;)
Very timely response.   I was having this exact issue a moment ago.

domesticatedviking, 2019-06-11:
Want to leave this great gist here - I found it very helpful getting the Mac version to compile.  https://gist.github.com/fujin/78e3ebda0f6c39d23ce1bb2a904db83c