prusa3d/PrusaSlicer #2114: [Feauture request] SLA Anti-Aliasing option

fookatchu, 2019-04-14:
It would be great if we could select the amount of Anti-Aliasing for the generated images or disable it completly. Currently Slic3r PE will always generate 8-Bit Greyscale Images. For different resins it would be good to let the user have the possibility to tune or at least disable this.

bubnikv, 2019-04-14:
The upcoming beta2 provides a Gamma parameter. Set it to 1 -> your raster
will be black and white, thresholded at a 1/2 of the intensity. Set it
above 1 to correct for resin non-linearity. Please share your results.

fookatchu, 2019-04-15:
Exactly what I wanted to hear, but I should have waited a day before posting ;). I tested this with beta2 and can confirm that it works like you described.

closed 2019-04-15 by fookatchu

bubnikv, 2019-04-15:
Are you getting any useful results from it? What gamma value works for you
and for which resin?

fookatchu, 2019-04-15:
I have no test prints until now. I just analyzed the resulting images. I asked for this as I have heart that anti-aliasing can have impacts beyond smoothing on the curing of some resins. I am mostly interested in precision versus nice looks and wanted to analyse if there is something true to those rumors. I hope I can disprove those. Also it is generally nice to have more options for tuning if necessary. 
As a side note: I have no SL1 yet, but have used the *cough* Photon to test Slic3r PE's mSLA features and I am very happy with it at the moment.

bubnikv, 2019-04-15:
>  I have no SL1 yet, but have used the *cough* Photon to test Slic3r PE's
mSLA features and I am very happy with it at the moment.

We would be happy, if you could please write down the pros and cons of the
two softwares when used with the Photon.

fookatchu, 2019-04-15:
The tl;dr would be that the Anycubic Photon Slicer (which essentially is the Chitu Slicer) is not open source and therefore not acceptable for me if there is an alternative. The interface is also lacking a little eye candy and usabillity in comparison to the new Slic3r PE interface. I would rather not comment on the support generation, as I have too limited experience with this at the moment, but it was ok.

The new version of their Slicer (ChituBox) improves somewhat on features with the 'hollow model' and 'make hole' functions, which drops the need to prepare this with MeshMixer. This is definitly a pro, but I have not used these features yet. I encountered some bugs when adding manual supports, where the supports where not attached to the model but out of the print area. 

Slic3r PE is currently not able to export to .Photon files, which is why I wrote a converter for this. This is currently a personal project largely based upon the work of [PhotonFileEditor](https://github.com/Photonsters/PhotonFileEditor), but with a rewritten code base. Once I tested this a little more I will make it available on GitHub. This is currently the biggest downside, obviously. Another downside (was) that the Photon does currently not support anti-aliasing, which is why the conversion from .SL1 to .Photon dropped some information and the resulting slices between the slicers would differ but only so much. As a bonus to my original obsolete feature request, Slic3r PE is completly 'image' compatible with the Photon even without AA. According to some people the Photon will get a new firmware which will enable support for anti-aliasing.
Pros of Slic3r PE are the user interface (gimbles), the 'find orientation' feature, and the support generation / modification. Again, I mostly used Slic3r PE for my Photon prints, which is why this opinion might be biased. 
My biggest Pro for Slic3r PE is that I mostly print on my MK3 and will do so in the future. I don't see a reason to use a different, closed-source software if I can use the same software for all my printers. It would be nice if Slic3r PE would add support for the Photon, but I honestly can understand that this will most likely not happen, if Josef has to pay developers for it ;).

bubnikv, 2019-04-15:
> It would be nice if Slic3r PE would add support for the Photon, but I
honestly can understand that this will most likely not happen, if Josef has
to pay developers for it ;).

Exporting a PNG archive in a slightly different format may be supported by
Slic3r in the future if it is not too difficult for us to integrate &
maintain. We do support other FDM printers indeed.

fookatchu, 2019-04-16:
The Photon file format is unfortunately not a simple Zip-file like the SL1 format, but a custom binary structure with the images encoded as run-length-encoding. Nothing fancy, but also not trivial but already reverse engineered. I would love to see this implemented, so please feel free :). I would help, but my skills are mostly Python related...
I will try to release the converter soon so that other people can use Slic3r PE with this.

bookkojot, 2019-06-07:
Currently I working on some tools for Photon printer, main problem for such converters: lack info about layers size, exposure time, offtime, distance of each retract and so on. Otherwise, you can export your slices as movie (gif, even mp4/avi) and convert to Photon with simple tool (I wrote this in 1 day).

fookatchu, 2019-06-07:
I wrote the mentioned tool and the necessary lib some time ago but forgot to mention it here: https://github.com/fookatchu/SL1toPhoton