prusa3d/PrusaSlicer #2115: Support for Junction Deviation

guestisp, 2019-04-14:
Newer marlin version removed the Jerk support in favor of Junction deviation, thus the `M205 X Y Z` can be replaced with `M205 J`

A new value in the machine limits should be added, for setting Junction Deviation. If that is set, the jerk value should be removed from the gcode (they are ignored by the firmware)

Adrian-at-CrimsonAuzre, 2019-04-23:
Piggybacking off of this, could we have options for Jerk, Junction, and None?
Klipper does not support Jerk control and I would like to be able to disable it so it doesn't affect print ETAs and generate unnecessary gcode.

supermerill, 2019-04-25:
But older version of marlin, installed in many chinese (and the prusa fork in my mk2) is using only the jerk setting.

madsi1m, 2020-04-04:
Maybe do the default thing of having the value 0 as "use default from firmware" for jerk and add another line for junction deviation which can also be disabled with 0

supermerill, 2020-10-13:
How to support old marlin and newer marlin? outputting both?

madsi1m, 2020-10-13:
I know "new marlin" ignores the old commands, not sure going the other way.

I too have since moved to Klipper and would like a None option. Or do it how SuperSlicer does it and add a Klipper gcode selector in printer settings

ManuelMcLure, 2020-10-14:
Unfortunately, even with "new Marlin" there are various possible scenarios, since you can disable Junction Deviation and go back to classic jerk when configuring the firmware. These are the possible options (as of this writing):
Configuration|Options to `M205`
---------------|--------------------
Classic Jerk without Linear Advance|X Y Z E
Classic Jerk with Linear Advance|XYZ
Junction Deviation or without Linear Advance|J

I believe Marlin will just ignore any options that are not relevant to the configuration.

vovodroid, 2024-01-11:
It seems quite easy to support correct time estimation for JD (Junction Deviation). Slicer code uses acceleration and jerk values for calculations, but if I'm correct JD is produced from acceleration and jerk, thus jerk could be calculated from JD and acceleration.

According to https://blog.kyneticcnc.com/2018/10/computing-junction-deviation-for-marlin.html

[image]

so
`
const float axis_max_jerk = sqrt(jd/0.4*acceleration);
`
instead of
`const float axis_max_jerk = get_axis_max_jerk(static_cast<PrintEstimatedStatistics::ETimeMode>(i), static_cast<Axis>(a));`

in `void GCodeProcessor::process_G1`

Is it correct?