prusa3d/PrusaSlicer #2116: Slowdown speed between fast travel moves

guestisp, 2019-04-14:
Let's assume a huge layer (so the small perimeter speed won't apply) but made with multiple very small parts, in example, 3mm diameters circles.

If you have tens of these in the same layer, the small perimeter speed isn't applied, because the overall layer radius is greater than 6.5mm

But printing these small circles at full speed isn't good, so I propose to add a new setting for slowdown the speed is 2 travels are made in less than X seconds. This means that a very small part is printed.

I hope is clear what I mean.

Sebastianv650, 2019-04-14:
I think you have a misunderstanding of the small perimeter speed. This isn't applied based on the overall printed area, but on the length of the perimeter which is actualy printed. So if you have a ton of 3mm diameters all over the build plate, they will all print with small perimeter speed.

guestisp, 2019-04-14:
I'm not really sure. I'm printing some very small circles too fast. I've set small perimeter speed at 20 and perimeters 60 but is not printing at 20 for sure. Much more near 60.

Sebastianv650, 2019-04-14:
Just upload the gcode so we can have a look and also attach the 3mf project. You can also use gcode.ws to check the speeds yourself.

guestisp, 2019-04-14:
Based on gcode.ws speed is correct
Anyway, can you confirm that small perimeter speed is used when diameter of each single part is less than 6.5mm and not for the whole layer?

If that's the case, i can close this useless issue

Sebastianv650, 2019-04-14:
The tooltip states a radius <= 6.5mm. So this should equal a trigger length of the perimeter of ~41mm.
Yes, it's definitly not the whole layer. The whole layer speed (based on area) can only be influenced by the cooling logic.

closed 2019-04-14 by guestisp