prusa3d/PrusaSlicer #2117: Zero infill and <= 2 perimeters forces physically impossible extrusion

Ianmcmill, 2019-04-14:
### Version
1.41.3+win64
+
Repetier-Host 2.1.3

### Operating system type + version
Windows 10 

### 3D printer brand / version + firmware version (if known)
P3Steel Toolson
Marlin 1.1.9

### Behavior No. 1
When slicing with the following settings, sliced perimeters become bloated and required amount of filament is affected by factor *10.
Infill = 0%
Perimeter <= 2
Bloated perimeters are printed as seen in the preview.
Extruder stalls because of insanly high value for extrusion.
In other words:
Extruder tries to extrude lines with a thickness beyond the physical possible.

When slicing in Slic3r's built-in plater, this does not occur. Only in Repetier-Host 2.1.3.

Not sure if this is a Repetier-Host problem or Slic3r PE slowly becoming incompatible with RH.

Screen-cap

[image]

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

Gcode
[20mm-box.txt](https://github.com/prusa3d/Slic3r/files/3077568/20mm-box.txt)

bubnikv, 2019-05-04:
Repetier communicates with Slic3r through an .AMF file. You should be able to send us the particular AMF file. Please contact Repetier support as to how to get the AMF file.

rtyr, 2021-04-22:
No response. Closing.

closed 2021-04-22 by rtyr