prusa3d/PrusaSlicer #2120: Brim generation problem with certain .stl files

sonofsignalman, 2019-04-15:
### Version
Version 1.41.3+win64

### Operating system type + version
Win10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Slic3r Prusa Edition
Version: 1.41.3+win64
Build:   Slic3rPE-1.41.3+win64-full-201902111723
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: 
Microsoft Windows [Version 10.0.17134.81]

Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=MSWin32, osvers=10.0, archname=MSWin32-x64-multi-thread
    uname=''
    config_args='undef'
    hint=recommended, useposix=true, d_sigaction=undef
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=undef, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cl', ccflags ='-nologo -GF -W3 -O1 -MD -Zi -DNDEBUG -GL -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY',
    optimize='-O1 -MD -Zi -DNDEBUG -GL -fp:precise',
    cppflags='-DWIN32'
    ccversion='18.00.40629', gccversion='', gccosandvers=''
    intsize=4, longsize=4, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=undef, longlongsize=8, d_longdbl=define, longdblsize=8, longdblkind=0
    ivtype='__int64', ivsize=8, nvtype='double', nvsize=8, Off_t='__int64', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='link', ldflags ='-nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'
    libpth=\lib
    libs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    perllibs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    libc=msvcrt.lib, so=dll, useshrplib=true, libperl=perl524.lib
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_win32.xs, dlext=dll, d_dlsymun=undef, ccdlflags=' '
    cccdlflags=' ', lddlflags='-dll -nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'

  @INC:
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\5.24.0\MSWin32-x64-multi-thread
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\5.24.0
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\MSWin32-x64-multi-thread
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5
    C:/PROGRA~1/Prusa3D/Slic3rPE/lib
    C:/Program Files/Prusa3D/Slic3rPE/lib
    .
OpenGL installation
GL version:   4.3.0 - Build 20.19.15.4531
Vendor:       Intel
Renderer:     Intel(R) HD Graphics
GLSL version: 4.30 - Build 20.19.15.4531
Installed extensions:

GL_3DFX_texture_compression_FXT1
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
[... 190 lines cut ...]
Prusa i3 Mk3

### Behavior
* with certain .stl files only part (1/2?) of the brim is generated or perhaps the brim is offset
* _Steps needed to reproduce the problem_
  *.stl file loaded:

[Middle.zip](https://github.com/prusa3d/Slic3r/files/3079362/Middle.zip)
check Brim option and slice.

* _Expected Results_
Preview and printing should show base of object surrounded by brim
* _Actual Results_
Brim surrounds only half the base of object (both in preview view and when printed.
Work around - set a wider brim
5mm brim -> [image]
20mm brim->[image]

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_
[Middle_0.15mm_PLA_MK3.3mf.zip](https://github.com/prusa3d/Slic3r/files/3079410/Middle_0.15mm_PLA_MK3.3mf.zip)

neophyl, 2019-04-15:
Just downloaded your project and as I thought the bottom of your model isnt placed onto the bed.  If you slice it and check from under the bed you will see it slopes upwards.  The brim is only put around the parts touching the bed on the bottom layer.

If you use the place to bed function and then reslice you will see the brim correctly surrounds the model.

sonofsignalman, 2019-04-15:
Thank you. Tim

closed 2019-05-04 by bubnikv