prusa3d/PrusaSlicer #2122: Support enforcer - box or cube?

espr14, 2019-04-15:
1.42.0 b2 win64

I cannot set different XYZ size dimensions for support enforcer box. If I set one, it will change the others to the same value. So it's always a cube.

neophyl, 2019-04-15:
I've just tried with the beta 2 and I can change the size of a support enforcer modifier.  You can either use keyboard 'S' to bring up the bounding boxes which depending on which you select will change individual axis and allows on screen dragging or from the right hand listing you can type in a % or an actual dimension.  However if you type in a % don't forget to first untick the Scale factor padlock (orange icon goes grey when unlocked) otherwise they are all linked.

espr14, 2019-04-15:
Thanks a lot. I forgot that there's a lock. However, it is similar to locks in settings but they do something else so it's confusing. Moreover, a lock is more to connected to fixed values which cannot be changed. Also the lock is next to scale factor, not next to size setting. Looks like it should be re-thinked. I'm not an UX designer so I have no clue now.

espr14, 2019-04-16:
However, the lock should be opened by default.

Mirarkitty, 2019-04-21:
I agree the lock should be opened by default on modifiers. On models it makes sense to have the lock on.

bubnikv, 2019-05-04:
thank you for your suggestion.

closed 2019-05-04 by bubnikv