prusa3d/PrusaSlicer #2125: Cooling of support interface

espr14, 2019-04-15:
1.42.0 b2 win64

There is settings for fan speed general and for bridges and overhangs. It would be nice to control fan speed for support interface as well. 

If I turn off the fan for the printed object to increase blending of layers, it could blend with support as well. So I would like to set fan speed to 100% for all support interfaces to avoid blending. But it should be 0% for support infill not to cool down the printed object around.

bubnikv, 2019-05-04:
You can prototype your very unusual idea by writing a post processing script based on the extended G-code annotation Slic3r is able to generate.

wavexx, 2020-10-08:
I second this.

To rephrase: allow control of the fan speed as a seperate setting while laying down the first interface in contact with the model. It shouldn't be changed for any other scenario (such as interface on top of support).

The idea is to reduce adhesion of the first interface with the part.

I tested this idea on TPU using a post-processing script. The difference it does is significant: the interface changes from "barely less welded" (which with TPU means: a blade is required) to "actually detachable just using pliers!" using 100% fan.

TPU is perhaps the best scenario for this setting: no fan is usually required and doesn't warp either.
But I suspect PETG would also benefit for the same reasons! (I'll try that soon)

HasslerEngineering, 2020-10-14:
I trice this. 
printing PETG with support is just a pain and this feature would make it a little better.

Sleslie706, 2020-11-19:
Yes I support this too.  I am also finding it difficult to remove bottom support interfaces when printed on print surfaces, particularly for PETG.  It appears that the fan setting for the main print (eg no fan for first layers) is being applied to the bottom of support interfaces even when the interface is on printed surfaces (I assume because the first interface layer, wherever it appears, is considered the first layer for the support).  A setting for fan speed for initial layer of support interface when printed on object would be very helpful (initial support interface levels on the build plate could be treated the same as the main print object).

leonhma, 2023-06-24:
+1 (see @wavexx's comment). Cura has had this feature for some time now. I just recently switched since cura doesn't work on my chromebook and this is one of the biggest pain-points I have with PrusaSlicer

YosypC, 2024-07-25:
Why has this marked with "very unusual idea"? This is a needed feature!

Tupson444, 2024-07-25:
Correct me if I'm wrong, but even without knowing code I can see that this should be very easy to implement, as it's just turning fan on to 100% when starting first layer after support, and switching back to whatever fan speed was before that. Therefore really no need to mark it as "very unusual" and wait for "experimental testing", as it can be implemented with no fuss.

SarraceniaWilds, 2024-10-01:
Please remove the very unusual label, this is not uncommon or unusual and a very useful setting used by many in other slicers.

Tupson444, 2026-09-13:
still wanted

Jan-Soustruznik, 2026-09-30:
Hello, @Tupson444 , thank you for update, According to new [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I closing this feature request and converting it into a Discussion thread to hear what the community thinks about. Thank you, and have a nice day