prusa3d/PrusaSlicer #2126: Mirror

Mack1945, 2019-04-15:
### Version
_Version of Slic3r Prusa Edition used goes here_Slic3r Prusa Edition 1.42.0-beta2 and 1.42.0-alpha7, 

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version Windows 10
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) Prusa MK3 with MMU2
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ Prusa MK3 with MMU2, printer FM= 3.7.0-2201, MMU2 FW 1.0.5-297

### Behavior
* _Describe the problem_ No way to mirror an complete object
* _Steps needed to reproduce the problem_ put an object on the bed no way to mirror it
  * _If this is a command-line slicing issue, include the options used_No
* _Expected Results_ right hand object to left hand object
* _Actual Results_none could not mirror
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_YES, unless I have over look the way to mirror an object

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

Mack1945, 2019-04-15:
Slic3rPE-1.40.1.win64-full-201807051331 and Slic3rPE-1.41.3+win64-full-201902111723 you could mirror an complete object

bubnikv, 2019-04-15:
There is an option at the pop up menu that opens if you click the right
mouse button over an object in the 3D scene.

Mack1945, 2019-04-15:
Thank you for the help

closed 2019-04-24 by lukasmatena