prusa3d/PrusaSlicer #2127: [Feature Request]

Jacky2k, 2019-04-15:
It would be great if Slic3r would calculate the amount of time the printer speed is limited by a certain parameter. Giving this in a small list would make it much easier to figure out what speed limit is the most limiting for the current configuration. This way it would be easy to optimize a profile for speed and get a feeling for what is going on in the background.

bubnikv, 2019-04-15:
How about this?

[image: image.png]

Jacky2k, 2019-04-16:
Maybe I am too stupid, but where can I see the image? The is no link, no image displayed...

bubnikv, 2019-04-16:
sorry, I suppose I was impatient and the image did not get uploaded.
[image]

Jacky2k, 2019-04-16:
OK, I am not 100% sure what these values mean. Correct me if I am wrong: Obviously the volumetric flow rate is not optimal with 1.51 mm³/s when printing the first layer, as it also could be 15 mm³/s. I guess it is limited by some other value (e.g. the max. perimeter velocity) but this fact is not shown. Also what if the first layer takes only 5 minutes to print, while the rest takes 500 minutes? This would make the first layer speed more or less irrelevant to the total print time.

Mirarkitty, 2019-04-21:
It also doesn't show mm/s extruder motion speed.

bubnikv, 2019-05-04:
ok, the hint I directed you to has a different purpose. It's purpose is to make sure the maximum volumetric speed of your extruder is not exceeded.

It is quite difficult to provide such hints you are asking for per parameter. Cura provides time spent per feature (perimeter, infill, supports etc). That may help to some extent and it is not too difficult to implement.

guestisp, 2019-05-04:
> Cura provides time spent per feature (perimeter, infill, supports etc). That may help to some extent and it is not too difficult to implement.

Very interesting

rtyr, 2021-01-28:
Implemented in 2.3.0. Closing.

closed 2021-01-28 by rtyr