prusa3d/PrusaSlicer #2128: Can no longer enter infill values manually

NovaViper, 2019-04-16:
### Version
Slic3r 1.42 Beta 2

### Operating system type + version
Windows 10 1809

### 3D printer brand / version + firmware version (if known)
Ender 3 (Mod

### Behavior
* _Describe the problem_
Since the new update, I notice that I can't seem to enter in values via typing them in the either infill settings within Print and in the main menu
* _Steps needed to reproduce the problem_
1. Type in a random number in the infill settings
2. Notice how it never seems to register the change
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Settings to change automatically
* _Actual Results_
Nothing happens, the other settings entered using the pop up menu are still being used
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

th33xitus, 2019-04-16:
I can confirm this.
I tried to set 15% infill manually to 18% infill and nothing changed. Usually you have to re-slice the object.
Only setting it to 20% or 10% registers a change and force you to re-slice.

It would be nice being able to set 0-100% in 1% increments (at least manually)

bubnikv, 2019-04-17:
This was a Windows specific bug in wxWidgets, fixed with
https://github.com/prusa3d/wxWidgets/commit/5aa36913873f5f3cd02a81f3927078a9db9d1b1d

We fixed that at our build server, so it will be part of the next release.
Closing

closed 2019-04-17 by bubnikv