prusa3d/PrusaSlicer #2129: Hole in print

my3thumbs, 2019-04-16:
### Version
Slic3r Prusa Edition 1.41.3+

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
macOS Mojave version 10.14.4
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_Prusa i3 MK3 / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* I open the STL file below, slice it and there is a hole in the center of the body section behind the feet. it is in line with the key chain hole horizontally. 
* Flexi Rex with hole for keychain (https://www.thingiverse.com/thing:2776182) needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* The item should have no hole in that area
* _it prints as the slicer shows with no PLA in that area (hole) so the print fails every time_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?- No I think it is a bug?_

#### Project File (.3MF) where problem occurs
flexi-rex_hidden_0.15mm_PLA_MK3.stl a Slic3r PE Project File (flexi-rex_hidden_0.15mm_PLA_MK3.3mf)

bubnikv, 2019-05-04:
I tried to slice these files at 0.15mm layer height for the I3 MK3 printer and I see no issues. Screenshots and a 3MF file would help.

rtyr, 2020-05-04:
I was able to reproduce it, but it is caused by broken model. Right-click on the icon will fix the issue. Closing.

[image]

closed 2020-05-04 by rtyr