prusa3d/PrusaSlicer #2131: Menu kezboard shortcuts still wrong

tkapler, 2019-04-16:
[###](url) Version
Version 1.42.0-beta2+win64
### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
not relevant
### Behavior
Menu keyboard shortcuts still shown wrongly in View menu (in previous alpha versions it was at other menus too)
[image]

bubnikv, 2019-04-17:
this is interesting, I don't see it. Do you get it even if you switch the
language to English?

bubnikv, 2019-04-17:
Sorry, I see now it is English. Do you see the issue in other menus as well?

bubnikv, 2019-05-04:
Is it possible, that you copied new slic3r over the installation directory of the old slicer, so you mixed language dictionaries of different versions of Slic3r?

bubnikv, 2020-03-26:
No response, closing.

closed 2020-03-26 by bubnikv