prusa3d/PrusaSlicer #2132: Elephant foot compensation 0.2 issue

espr14, 2019-04-16:
1.42.0 b2

Use the following STL file, set 0.2 Quality profile, and set "Elephant foot compensation to 0.2". It will create first layer as shown in the image below. If you set 0.1 or 0.3 it will do what would you expect.

[el02.zip](https://github.com/prusa3d/Slic3r/files/3085445/el02.zip)

[image]

zehkaiser, 2019-04-21:
I also have this issue when using a multi material object. The preview will show a solid face (made up of different objects/colors), but when slicing it will add the .2mm elephant foot compensation between all objects, causing there to be gaps in the surface.

bubnikv, 2020-01-02:
The elephant foot compensation improvements are part of PrusaSlicer 2.2.0-alpha2. Closing.

closed 2020-01-02 by bubnikv