prusa3d/PrusaSlicer #2133: First bridge layer infill density

espr14, 2019-04-16:
1.42.0 b2

Threads in wide bridges are shrinking unevenly and they tend to stick together. This creates weird random patterns. If I modify my model so it creates 50% 1st bridge layer infill density, it looks much better. However, it requires to modify every bridge with known layer height apriori. So it would be nice to have "1st bridge layer density" setting to change this for whole model (or just in modifier area).

100% bridge infill:
[image]

sharktank72, 2020-09-17:
Not so unusual, I'd really like this feature too.  It helps keep the bottom of the object really clean and supports to release better.

espr14, 2020-09-17:
It's already implemented in SuperSlicer as `bridge_overlap`.

jeremy-asher, 2026-09-18:
It would be great to see this implemented.