prusa3d/PrusaSlicer #2134: Feature request: make object labels match instance names

paulr4, 2019-04-16:
### Version
1.42.0-beta2+win64-full

### Operating system type + version
Windows 10

### Behavior
Slic3r's object labelling feature works well with 'Octoprint-Cancelobject'. However, one small improvement would improve usability.

Slic3r currently gives objects a unique label in the gcode file. For multiple instances of a single object, the label could be something like 'test.stl id:0 copy n', where n is the copy number, starting from 0. Slic3r identifies multiple instances differently, with an instance number starting from 1.

Would it be possible to use the same numbering system for both purposes? I ask because the method for selecting the object to cancel in OctoPrint can be inaccurate, and a way to double-check the object number to be cancelled against the Slic3r job would help avoid mistakes.

bubnikv, 2019-05-04:
it certainly makes sense. It requires a bit of gymnastics though, as the slicing back-end currently is not aware of possible instances, which are out of the print bed, therefore not printed. But it should be possible with some effort.

paulr4, 2019-05-04:
I guess an instance/object name would need to be passed to the slicer backend along with the shape data. Instances off the print area wouldn't matter since they don't need to be named, sliced, or cancelled.

robot256, 2023-04-12:
This feature is present in Prusa Slicer 2.5.0 at least.  It works with Klipper "exclude object" functionality as well.

It is broken in 2.6a6.  The copy ids are assigned randomly or all zero. See https://github.com/prusa3d/PrusaSlicer/issues/10257

kubispe1, 2023-05-17:
Thanks, @robot256. There is an actual statement from the developer regarding this topic in PS 2.6.0-beta1.
https://github.com/prusa3d/PrusaSlicer/issues/10257#issuecomment-1505681457

closed 2026-09-29 by prusaslicer-bot[bot]