prusa3d/PrusaSlicer #2135: Inherited printer profiles

guestisp, 2019-04-17:
Is possible to create some printer profile that inherit settings from a defined profile ?
I don't see how to change the "default" profile and how to create new profiles bound to a specific profile.

In example: I have multiple print beds (some glasses, some magnetic and so on). Based on the bed I currently use, I have to use a different start gcode to load the proper mesh before bed levelling. Currently I had to create about 4-5 printer profiles that I have to keep in sync on every change.

bubnikv, 2019-04-17:
@guestisp @espr14 would you guys please search the issue tracker before issuing questions / proposals here? It really puts a lot of burden to do the homework for you.

guestisp, 2019-04-17:
I've searched. There isn't anything useful with "inherit" regarding this possibility to inherit profiles via the GUI

espr14, 2019-04-17:
@bubnikv There are lots of issues without any label. I cannot even add one so it's hard to find something specific. Yet, I haven't add proposal which was added before.

Inherited profiles would be good but it would require some UX/UI rebuild.

supermerill, 2019-04-17:
To create inherited profile, look at PrusaResearch.ini 
It's relatively easy to reverse-engineer from that.

guestisp, 2019-04-17:
I know how to do from ini files. I was looking for something in the GUI.
Is this currently not possible from the GUI ?

vojtechkral, 2019-04-18:
> Is this currently not possible from the GUI ?

It's not exposed in the GUI for user-defined presets right now. Of course, yes, it could be exposed and the GUI for it could be created, but no, it won't be done any time soon because I don't think anyone has the time & energy to do it right now.

bubnikv, 2019-05-04:
Duplicate of https://github.com/prusa3d/Slic3r/issues/1108
Closing.

Thank you for the homework. It is a rainy Sunday 16:10 here in Czech Republic.

closed 2019-05-04 by bubnikv