prusa3d/PrusaSlicer #2136: GUI Flashes and shows layer view briefly after every move.

jweaver100, 2019-04-17:
### Version
1.42.0 Beta 2 (Same thing happens on all 1.42 Alpha and Beta versions)

### Operating system type + version
Windows 7 (Build 7601, SP1) - 6 Bit

Output of System Info:

Slic3r Prusa Edition
Version:   1.42.0-beta2+win64
Build:     Slic3rPE-1.42.0-beta2+win64-full-201904140830

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 7 (build 7601, Service Pack 1), 64-bit edition

OpenGL installation
GL version:   4.6.13547 Compatibility Profile Context 25.20.15027.9004
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon HD 7700 Series
GLSL version: 4.60
Installed extensions:
GL_AMDX_debug_output
GL_AMD_blend_minmax_factor
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_sample_positions
GL_AMD_gcn_shader
GL_AMD_gpu_shader_int64
GL_AMD_interleaved_elements
GL_AMD_multi_draw_indirect
GL_AMD_name_gen_delete
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_atomic_counter_ops
GL_AMD_shader_stencil_export
GL_AMD_shader_stencil_value_export
GL_AMD_shader_trinary_minmax
GL_AMD_sparse_texture
GL_AMD_stencil_operation_extended
GL_AMD_texture_cube_map_array
GL_AMD_texture_texture4
GL_AMD_transform_feedback3_lines_triangles
GL_AMD_transform_feedback4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_parallel_shader_compile
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_query_buffer_object
GL_ARB_robust_buffer_access_behavior
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counters
GL_ARB_shader_ballot
GL_ARB_shader_bit_encoding
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_image_samples
GL_ARB_shader_texture_lod
GL_ARB_shader_viewport_layer_array
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sparse_buffer
GL_ARB_sparse_texture
GL_ARB_spirv_extensions
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_snorm
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_draw_buffers
GL_ATI_envmap_bumpmap
GL_ATI_fragment_shader
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_buffer
GL_EXT_copy_texture
GL_EXT_depth_bounds_test
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_histogram
GL_EXT_memory_object
GL_EXT_memory_object_win32
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_semaphore
GL_EXT_semaphore_win32
GL_EXT_separate_specular_color
GL_EXT_shader_image_load_store
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texgen_reflection
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer_object
GL_EXT_texture_compression_bptc
GL_EXT_texture_compression_latc
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_EXT_vertex_attrib_64bit
GL_IBM_texture_mirrored_repeat
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KTX_buffer_region
GL_NV_alpha_to_coverage_dither_control
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Prusa MK3S - FW 3.7.0 

### Behavior

In all versions of 1.42 I see a graphical problem that was not in 1.41 or any earlier version

The issue manifests iteself as a white 'flash' every time you add a new item, move a model or change a config value...

I have captured a record here:

https://youtu.be/rW9ovciO8TQ

And after looking at this its clear that the flash is caused by the GUI briefly showing the 'layer view' after every move.

I have tested this on other PCs (Windows 7 and 10) and cannot reproduce this issue.. So this appears to be something specific to my PC.. Its hardly a big issue, but is really annoying and I hope something can be done to fix it.

#### Project File (.3MF) where problem occurs

Simple add any STL and move it, and the problem occurs.

bubnikv, 2019-04-17:
I have never seen something like this. I wonder whether it is a Windows 7
issue, issue of your graphics card, or a bug in Slic3r.

You may try to copy the opengl.dll from the mesa directory one level up, or
to start Slic3r 1.42.0-beta2 with a --sw-renderer parameter to switch to a
software renderer, I wonder whether you will still have this issue. Please
understand, that with a software renderer Slic3r will run much slower.

bubnikv, 2019-04-17:
Would you also please try to update your graphics display?

jweaver100, 2019-04-17:
Sorry.. Should have said.. I had already upgraded the graphics, but last night did it again just to be sure, to the latest drivers (1st April 2019)..

I am running an ATI Radeon HD 7770 (pretty average performance at best) and am running 19.3.3 Drivers.

I will try your other suggestions later when i get home.

jweaver100, 2019-04-17:
YOU ARE AWESOME....... I just connected via Teamviewer and moved the "opengl.,dll" file up one level and now the flickering is gone....  

Was that just a lucky guess.. Or is there a reason why this might help?

jweaver100, 2019-04-17:
It looks like this change "downgraded" OpenGL.. Now I have copied the file its:

OpenGL installation
GL version:   3.0 Mesa 12.0.0-rc2
Vendor:       VMware, Inc.
Renderer:     Gallium 0.4 on llvmpipe (LLVM 3.6, 128 bits)
GLSL version: 1.30

Where as before it was 

OpenGL installation
GL version: 4.6.13547 Compatibility Profile Context 25.20.15027.9004
Vendor: ATI Technologies Inc.
Renderer: AMD Radeon HD 7700 Series
GLSL version: 4.60

I guess after your change, I am not using the ATI Radeon varient of the OpenGL drivers?

bubnikv, 2019-04-17:
the advice I gave you effectively cripples down the graphics card, so this was not such an AWSOME advice, but a workaround. 

I will test on my two Windows 7 machines at home, but I have a feeling it will work fine. We have no explanation of what is going on on your computer.

jweaver100, 2019-04-17:
At the end of the day, if I find problems, I can always go back and live with the Flashing.. its not the end of the world, but I wanted to raise it just incase others have this or you had a solution...

Its odd as I use 100s of other applications and don't have problems with any others. Thats primarily why I havnt' upgrade my PC in years... It does everything I want.

jweaver100, 2019-04-20:
Just to give you an update.. I have been using this "crippled" mode for a few days and can't find ANY ill effects.. Slic3r is now running exactly as it does in 1.41 and below.. And I am still able to use the new features, so I am happy with this "solution".

jweaver100, 2019-04-21:
Actually I lied.. Everything seemed good until I played around with the 'layer view'.. In the "crippled" mode the layer view is unusably slow.. Where as when I delete the DLL and go back to default (where I have the flashing) the layer view is silky smooth again.

Looks like I am going to have to live with the flashing.. Its odd that it only started in 1.42.. Before that it was fine.

bubnikv, 2019-05-04:
Well we are lucky that you are the only one reporting such an issue.

You may try to go back in history and try various older 1.42 alpha builds whether anyone of them will work for you. This may give us a hint of what change in our code base triggers your issue. I am a bit pessimistic though, that we would be able to fix your issue without being able to put our hands on a similar hardware / software setup.

jweaver100, 2019-05-05:
Thanks for continuing to think about this one.. I realise this is impossible for you since I am on the only person who has reported this, but its odd that Slic3er is the only application I am using which isn't working perfectly.

This problem appeared in the first Alpha version (1).. I have run Alpha1, 4 and 7 (As well as the Betas) and they are all the same.. But I didn't see this in any of the previous releases, right up to 1.41.1

I recognise that this is a mega-low priority, but I am still hopeful that you manage to fix this one day, so if there is anything you want me to try or test I am happy to do what ever you ask.

bubnikv, 2019-05-06:
Thanks for testing. We did a huge leap between 1.41 and 1.42, a lot has
changed in the code.

jweaver100, 2019-05-09:
I just noticed something else about this problem which might be relevant... Have a look at this video:

https://youtu.be/6GYcAEw6RQM

After launching Slic3er, the behavour is as I described above (i.e I get a brief layer view every time I move)...

If I go into Print setting and change something and then back to plater.. Now after every move, I see the Print Settings screen.. And its the same for Filament Settings and Printer Settings.

Actually. I just tested again and I don't even have to change anything.. If I click on any of the 3 settings tabs and then back to plater, I see the last screen I was in after every object move.

I am not sure if this helps in any way, but I figured I would let you know.

pepelevamp, 2019-05-24:
> 
> 
> Thanks for continuing to think about this one.. I realise this is impossible for you since I am on the only person who has reported this, but its odd that Slic3er is the only application I am using which isn't working perfectly.

oh no you aren't baby! it happens to me as well. windows 7, nvidia 1030, driver 390.65. 
moving the opengl32.dll up from the mesa dir does fix it, but you also get a FPS drop.
and sure enough, putting opengl32.dll back into the mesa directory makes it flash again while being faster.

of course none of it matters because i wont be using this until _USB printing returns_ (hint)

jweaver100, 2019-05-30:
I have just upgraded to PrusaSlicer 2.0.0 Final and the "flashing" problem is happening on my Laptop at work now. Its exactly the same as it is at home (i.e I briefly see the 'layer' view every time I move and item...

OpenGL info is as follows:

PrusaSlicer
Version:   2.0.0+win64
Build:     PrusaSlicer-2.0.0+win64-201905201652

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 7 (build 7601, Service Pack 1), 64-bit edition

OpenGL installation
GL version:   4.4.0
Vendor:       NVIDIA Corporation
Renderer:     Quadro K2000M/PCIe/SSE2
GLSL version: 4.40 NVIDIA via Cg compiler
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
[... 217 lines cut ...]
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_video_capture
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

jweaver100, 2019-05-30:
Tested on my colleagues PC and he DOES NOT see this.. So its only my home and work PC, which means that its something about me, or something that I have installed on BOTH PCs...

I use something called "Display Fusion" (a tool to help with dual monitors) and have this on both PCs and its always been in the back of my mind that this is the cause.. But my colleague has it too and he doesn't have this.. .So I can rule that out.

I can't think of anything else I have on BOTH PCs that could cause this.. many previous versions of Slic3r for sure.. And lots of versions of CURA... And Sketchup.. But thats about it...

bubnikv, 2019-06-03:
Could you try to de-install the "Display Fusion" thing for a test, please?
Are these laptops and are they equipped with dual graphics cards?

jweaver100, 2019-06-03:
Let me clarify the situation with Laptops and Display Fusion.. I have already tried de-installing and it makes no difference.. But also, my colleague (who has the same laptop) also uses Display Fusion and does not have this issue...

There are 3 PCs involved in this test

Home - Desktop PC - Windows 7 - ATI/AMD Radeon HD 7700 Series with 2 monitor ports... Tried with single and dual monitor (physically removed cables).. And also removed Display Fusion. - Always flashing

Work PC (Docked) - Laptop - Windows 7 - NVidia/Mobile Graphics Card - When docked has 2 DVI outputs from the dock - Always Flashing

Work PC (Not Docked) - Laptop - Win 7 - NVidia - Displaying on laptop screen - Flashing

Collague PC (Docked) - Laptop - Win 10 - Nvidia - 2 Display Ports on Docking station - NO FLASHING.

So there is no obvious pattern to this.. The only thing that jumps out is Win7 vs Win10, but I have also tried other Win7 machines and do not have the flashing.

Last week I uninstalled 30-40 apps/tools/drivers on my work PC that I also use at home.. And its made no difference.. I still get flashing.

I am not sure if this helps, but I hope it will rule out some things.

bubnikv, 2019-08-01:
@enricoturri1966 found a similar issue on one of his computers with NVidia graphics card. Please retest with the upcoming PrusaSlicer-2.1.0-alpha1, at least on his computer it works.

jweaver100, 2019-08-01:
That is weird.. out of nowhere, i suddenly remembered this problem/thread and wondered if i would ever get a solution, 5 mins later i get your post..

I will gladly try this and will update you later.....

jweaver100, 2019-08-12:
Just had chance to test the 2.1.0 Alpha.. And I am happy to confirmed its fixed.. Good work....

bubnikv, 2019-08-19:
@jweaver100  cool, thanks. Closing.

closed 2019-08-19 by bubnikv