prusa3d/PrusaSlicer #2137: Optimize solid infill of narrow spaces

espr14, 2019-04-17:
1.42.0 b2

Solid infill changes two directions. But sometimes one direction causes a lot of turnarounds. This slows down the speed of print and the printer starts to vibrate a lot.

I propose to enable variable solid infill angle which changes angle according to filled space. If the turnarounds are too close (some threshold) then rotate the infill by a specified angle (to be set by user, typically 45°). It could use clock and anti-clockwise direction. If the result is better than the original then keep the new one.

This wouldn't apply for the first/last X (user defined) layers. On the other hand, this could also apply for narrow spaces which are filled along. There is so little space so 2 threads are too much thus 1 thread has to be used. In case of variable angle, it could be filled much better.

[image]

[image]

[image]

espr14, 2019-04-21:
#1531

bubnikv, 2019-05-04:
also similar to https://github.com/prusa3d/Slic3r/issues/28

closed 2026-09-29 by prusaslicer-bot[bot]