prusa3d/PrusaSlicer #2139: Toolchange load feedrate changes with loading distance

NielschB, 2019-04-18:
### Version
Slic3r 1.42.0-beta2+win64

### Operating system type + version
Windows 7 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 FW 3.7.0 / MMU2 FW 1.0.5

### Behavior
When changing "Extra loading distance" under printer settings in slic3r, the generated Gcode shows a change in feedrate on the toolchange part where the MMU and the extruder are both feeding the filament. The feedrate should be equal for both (1140 / 19.02 mm/s) and in my opinion not change when the loading distance changes. (Would be nice to make the complete toolchange gcode customizable from within slic3r)

bubnikv, 2019-05-04:
@lukasmatena what do you think?

lukasmatena, 2019-05-06:
@NielschB Sorry, I somehow missed the issue. There is a subtle bug regarding the feedrate calculation that I know of, but I doubt that this it the cause of your problem - I just checked what changing the 'Extra loading distance' does and it looks "almost" correct. Could you please post actual gcodes (and ideally a complete 3MF too) so I can see what differences are you talking about?

NielschB, 2019-05-07:
Hi, let me start from the beginning. I am in the process of "debugging" my mk3+mmu2 setup. The loading of the filament most of the times has some grinding issue by either the mmu2 gears or the bondtech gears, this I want to solve. So I started diving into the firmware and Gcodes to see what is going on. I found some parts that to my knowledge are not matching, but I could be missing some info.
All numbers that I will be using in the following explanation are with default slic3r / firmware / config settings : MK3 - 3.7.0 / MMU - 1.0.5 / Slic3r 1.42.0 B2 / Config 0.8.0 B3

Analysis of the initial loading:
A toolchange action in the Gcode triggers the mmu to load filament to the center of the bondtech gears (when calibrated the ptfe tube length correctly) and after that the mmu continues feeding the filament for 38.1mm @ 19.02mm/s (hardcoded) according to the mmu firmware. The extruder should feed with the same rate to prevent grinding.

Gcode shows the *initial* toolchange T[n] command followed by a G1 X55.0 E32.0 F1073.0.  Here I start wondering why the feedrate is 1073 and not 1141 to match the MMU. This part of Gcode is specified in Slic3r under printer settings > Custom G-Code> Start G-code. Additionally, why is the extrude length set to 32mm while the mmu is set to feed ~38mm? The next line of G code is G1 X5.0 E32.0 F1800. During the first 6mm of this gcode the mmu is still feeding at F1141, so that doesn't match either.

Analysis of the other toolchange loadings:
The other toolchange loads in the generated Gcode (Taken the striped lizard) show the following:
; CP TOOLCHANGE LOAD
G1 E14.4000 F1140
G1 X229.375 E50.4000 F1190
The first 14.4mm of extrusion is at the same feedrate as the MMU, but after that part, the remaining (38.1mm - 14.4mm) 23.7mm is at a faster feedrate (1190) but should be synced with the mmu.

When I change the loading distance in slic3r, the feedrate of F1190 changes, which I dont understand because it should be synced with the MMU for at lease another 23.7mm.

Now I found in slic3r the toolchain load settings under filament settings > advanced. Here the "Loading speed at the start" seems to correspond to the first line of gcode right after the toolchange, which is G1 E14.4000 F1140. I *think* this value should be valid for the length of filament the mmu is still feeding, which is 38.1mm instead of 14.4mm.  The value in the "Loading speed" field I cannot find in the generated Gcode, so I expect this value is not used correctly, have to test what happends when I change it.

I think this is enough info for now ;) I have added the lizard output gcode.

[3C_Lizard_1_0.2mm_PLA_MK3MMU2.gcode.zip](https://github.com/prusa3d/Slic3r/files/3153220/3C_Lizard_1_0.2mm_PLA_MK3MMU2.gcode.zip)

lukasmatena, 2019-05-07:
> why the feedrate is 1073 and not 1141 to match the MMU

I believe it might help to first see what feedrate actually means.

G1 X55.0 E32.0 F1073
G1 X5.0 E32.0 F1800

Let's have a look at the second move. It moves both X and E axes. In that case, the feedrate relates to the print head move. X moves from 55mm to 5mm, so it travels 50mm (absolute value of the difference). The feedrate is 1800mm/min=30mm/s, so the move takes 1.666 seconds. The extruder move is linear to move 32mm during that time, so its speed will be 19.2mm/s. If the feedrate was 1190, the e-velocity would be 12.69mm/s. The custom G-codes seem to be fine.

What is not correct is the wipe tower generated feedrate that you posted. Unless I made a mistake, the second command should be G1 X229.375 E50.4000 **F843** (to make the e-speed the 14mm/s set in Filament settings). You can try to change this feedrate to see if it helps with the grinding.

@akukan Can you please check that there is indeed a problem?

NielschB, 2019-05-07:
I see that my interpretation of extruder feedrate was all wrong, sorry about that! Now it all makes a bit more sense. The point that you mention about the feedrate on the wipe tower is one of the exact points where the grinding occurs. Also on the initial toolchange, but I will analyse that part again to check what's happening there.

Feeling a bit stupid and sorry for wasting your time. I do hope that the feedrate on the wipetower part *is* incorrect because that would probably solve one of my issues :)

Thank you for your explanation and analysis, I'll report back on the feedrate setting change of the wipe-tower.

lukasmatena, 2019-05-07:
>  I'll report back on the feedrate setting change of the wipe-tower.

Please, do.

> Feeling a bit stupid

Don't be. I'm fairly sure the part regarding the wipe tower will really turn out to be a bug. A bug caused by me and my inability to calculate it correctly too. So you can safely leave others to feel stupid.

Lukrative525, 2019-08-13:
I'm not sure if this is really on-topic here, but NielshB mentioned that the loading speed seemed like it wasn't being used correctly.

From the tinkering that I've done in the last few days, I have noticed this as well. I tested a bunch of different values, and then calculated the feeds out. Those calculated from the loading speed are off by a factor of 10, i.e., if 50mm/s is input as loading speed, the actual feed generated is 5mm/s.

I don't code, but I hope this is somehow useful.

lukasmatena, 2019-08-13:
@NielschB @Lukrative525 
Please, retest in PrusaSlicer 2.1.0-alpha1. The bug I mentioned above should be fixed there. It was a bit more subtle than forgetting to multiply by 10 somewhere, it related only to loading moves on the wipe tower where both X and E axis is moved and only in case that the X move was very short.

If you still feel there is something wrong, please enclose a 3mf with configuration and an excerpt from the gcode, so there is actually something to discuss about.

lukasmatena, 2019-10-01:
No more response - I assume the bug is indeed fixed.
Closing the issue - we can reopen it if needed.

closed 2019-10-01 by lukasmatena