prusa3d/PrusaSlicer #2140: "Layers" view is missing from 1.42.0 beta 2

GurliGebis, 2019-04-18:
### Version
1.42.0-beta2+Win64

### Operating system type + version
Windows 10 1803 64bit

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3, firmware version not relevant.

### Behavior
When I slice an object with 1.41 I have three tabs at the buttom ("3D", "Preview" and "Layers")
In 1.42.0-beta2 when I slice an object, I only have the option to switch between what would be "3D" and "Preview" - there is not option to switch to a view that shows single layers like in 1.41

#### Project File (.3MF) where problem occurs
Not relevant, since it happens with every object, from complex models to a simple 1x1x1 square :)

x0rtrunks, 2019-04-19:
Wasn't the layers view deprecated?

GurliGebis, 2019-04-20:
I haven't been able to find anything about it, which is why I'm creating this ticket. 🙂

I hope it isn't deprecated, since I'm using it often to evaluate if things are possible to print ect.

x0rtrunks, 2019-04-20:
From the 1.42 alpha 1 release notes
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha1

> 2D toolpath preview was dropped for now, however we will likely re-implement it as time allows. For now you can use the hot key "1" to orient the camera to look at the print bed from the top.

closed 2019-04-20 by GurliGebis