prusa3d/PrusaSlicer #2141: FeatureRequest: Automatic Load files <name>___supportblocker.stl and <name>___support.stl automatically if they exist for <name>.stl

ldpgh, 2019-04-18:
### Version
Slic3r Prusa Edition Version 1.41.1-rc+win64

### Operating system type + version
Feature should be OS idependent.

### 3D printer brand / version + firmware version (if known)
Feature should be printer idependent

### Behavior
If I load the file <name>.stl  it would be nice to load associated files
<name>___supportblocker.stl
<name>___support.stl
automatically if they exist.

<name>___supportblocker.stl ... contains figures to block any support.
<name>___support.stl ... contains figures to define a support region.

The figures are created in the 3D CAD (e.g FreeCAD). At the moment I keep all solids in the same FCStd-File and just export one after the other.

manticorp, 2019-04-21:
This is a great idea.

Alternatively, a way of packaging these files into a single file would be great.

ldpgh, 2019-04-22:
Packaging all solids into a single file seems to be possible ... see attached FreeCAD file and the exported STL-File ... [Test___slic3r_supportstructures_by_name___issue_2141.zip](https://github.com/prusa3d/Slic3r/files/3104711/Test___slic3r_supportstructures_by_name___issue_2141.zip)

But the handling within Slic3r is a challenge, because (at the moment) the entire STL-File is treated as a single solid.

If Slic3r splits the content into different part/modifier as it is the case if loaded from separate files it may be feasible.

For now the file-based proof-of-concept looks more simple to me. 
- There is no need to parse the STL-File:
- In addition any later modification changes the name of the new tree root-element. The original name need to be re-established afterwards manually.
- Though it seems to be the same for the file-based solution during the export. But once an older file is on the disc it is easy to (just) select the right file-name and press OK.

bubnikv, 2019-05-04:
> If Slic3r splits the content into different part/modifier as it is the case if loaded from separate files it may be feasible.

That certainly is possible, even with your version of Slic3r. It is the "Split part" button at the dialog you are showing at your screenshot.

Handling of parts is much easier with the current Slic3r 1.42.0-beta2. Did you try it? You have everything integrated in the main screen. Just click the "split to parts" button at the top toolbar and you may change the part purpose from "printable" to "modifier" or whatever you wish.

ldpgh, 2019-05-05:
... in response of @bubnikv 
Splitting the parts is possilbe as you mentioned ... 
[image]

... but it looses any information defining its role as associated by the name
[image]

The feature request behavior as described above shall remove/minimize any activities in case of many different solids needed to be loaded to the plate.

If there is better way to transfer those information from any CAD program to the slicer I can close this issue.

---

Btw ... have checked Slic3r 1.42.0-beta1 (beta2 not found) ... it starts with a simple issue ... 
   There is the menu from 1.41 to assign the modifier?
   Double click on the solid does not work.
I'm not amused with regard to the recommendation

---

I withdraw the feature request (no time for discussions) and close the issue.

closed 2019-05-05 by ldpgh