prusa3d/PrusaSlicer #2142: slic3r won't start so long as Fusion 360 is running

gnydick, 2019-04-18:
### Version
1.41.3+win64

### Operating system type + version
Windows 10 Pro x64 Version 1809 (OS Build 17763.437)
```
Slic3r Prusa Edition
Version: 1.41.3+win64
Build:   Slic3rPE-1.41.3+win64-full-201902111723
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: 
Microsoft Windows [Version 10.0.17763.437]

Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=MSWin32, osvers=10.0, archname=MSWin32-x64-multi-thread
    uname=''
    config_args='undef'
    hint=recommended, useposix=true, d_sigaction=undef
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=undef, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cl', ccflags ='-nologo -GF -W3 -O1 -MD -Zi -DNDEBUG -GL -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY',
    optimize='-O1 -MD -Zi -DNDEBUG -GL -fp:precise',
    cppflags='-DWIN32'
    ccversion='18.00.40629', gccversion='', gccosandvers=''
    intsize=4, longsize=4, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=undef, longlongsize=8, d_longdbl=define, longdblsize=8, longdblkind=0
    ivtype='__int64', ivsize=8, nvtype='double', nvsize=8, Off_t='__int64', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='link', ldflags ='-nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'
    libpth=\lib
    libs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    perllibs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    libc=msvcrt.lib, so=dll, useshrplib=true, libperl=perl524.lib
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_win32.xs, dlext=dll, d_dlsymun=undef, ccdlflags=' '
    cccdlflags=' ', lddlflags='-dll -nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'

  @INC:
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\5.24.0\MSWin32-x64-multi-thread
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\5.24.0
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\MSWin32-x64-multi-thread
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5
    C:/PROGRA~1/Prusa3D/Slic3rPE/lib
    C:/Program Files/Prusa3D/Slic3rPE/lib
    .
OpenGL installation
GL version:   4.6.0 NVIDIA 419.67
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2080 Ti/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
[... 366 lines cut ...]
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
```

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
If I have Fusion 360 running and start slic3r it just hangs after opening the window, but not showing any contents, just a white background. My guess is it's an opengl thing.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
N/A

bubnikv, 2019-04-19:
Please test with
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta2
the 1.41.3 will not be updated.

gnydick, 2019-04-19:
So far, I haven't been able to repro with my current version, as it doesn't always happen, but so far the beta works.

gnydick, 2019-04-26:
It's just repro'd with the original version reported, but Fusion isn't running. Here is the error from the windows event log.

Is there a way to enable debug logging in slic3r?

```
The program slic3r.exe version 1.41.3.0 stopped interacting with Windows and was closed. To see if more information about the problem is available, check the problem history in the Security and Maintenance control panel.
 Process ID: 24f4
 Start Time: 01d4fbf32673581b
 Termination Time: 19
 Application Path: C:\Program Files\Prusa3D\Slic3rPE\slic3r.exe
 Report Id: 2b67772a-4c26-4940-8f6c-36a944023f32
 Faulting package full name: 
 Faulting package-relative application ID: 
 Hang type: Unknown
```
```
- <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
- <System>
  <Provider Name="Application Hang" /> 
  <EventID Qualifiers="0">1002</EventID> 
  <Level>2</Level> 
  <Task>101</Task> 
  <Keywords>0x80000000000000</Keywords> 
  <TimeCreated SystemTime="2019-04-26T05:45:54.091846600Z" /> 
  <EventRecordID>23397</EventRecordID> 
  <Channel>Application</Channel> 
  <Computer>littlehoss</Computer> 
  <Security /> 
  </System>
- <EventData>
  <Data>slic3r.exe</Data> 
  <Data>1.41.3.0</Data> 
  <Data>24f4</Data> 
  <Data>01d4fbf32673581b</Data> 
  <Data>19</Data> 
  <Data>C:\Program Files\Prusa3D\Slic3rPE\slic3r.exe</Data> 
  <Data>2b67772a-4c26-4940-8f6c-36a944023f32</Data> 
  <Data /> 
  <Data /> 
  <Data>Unknown</Data> 
  <Binary>55006E006B006E006F0077006E0000000000</Binary> 
  </EventData>
  </Event>
```

gnydick, 2019-04-26:
OK, it's repeating with OP version, the beta you recommended is loading fine.

gnydick, 2019-04-26:
It looks like it has the problem loading when it tries to open on my second monitor. I did a full dump of the process and the output seem to indicate it was stuck somewhere around display detection, or something similar.

gnydick, 2019-05-01:
Yep, confirmed, it's while trying to open on the second monitor, I'll rename the issue.

bubnikv, 2019-05-04:
> Yep, confirmed, it's while trying to open on the second monitor, I'll rename the issue.

I am not sure I understand. Do you still have the issue with the 1.42.0-beta2?

gnydick, 2019-05-04:
No, it only happens with the OP version.

bubnikv, 2019-05-04:
thanks, I am closing this issue then, as we are going to release the 1.42.0 final soon.

gnydick, 2019-05-04:
Why closed, is this a known issue that's been fixed?

bubnikv, 2019-05-04:
@gnydick 
> The beta I tried is missing TONS of features,

which TONS?

> is that representative of the release version?

yes

gnydick, 2019-05-11:
Oh, I found them. There is the Simple Advanced Expert filters.

gnydick, 2019-05-27:
This shouldn't be closed, it's still happening on 2.0.0. I updated the original post with the right details.

bubnikv, 2019-06-03:
@gnydick 

> It looks like it has the problem loading when it tries to open on my second monitor. I did a full dump of the process and the output seem to indicate it was stuck somewhere around display detection, or something similar.

Would you please attach some information that indicates that it is stuck somewhere around display detection or something similar?

Don't you have a dual graphics card setup?

gnydick, 2019-06-04:
I can't upload the entire dump file, maybe this will help.

```
*******************************************************************************
*                                                                             *
*                        Exception Analysis                                   *
*                                                                             *
*******************************************************************************

KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.Sec
    Value: 1

    Key  : Analysis.Elapsed.Sec
    Value: 242

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 116

    Key  : Timeline.OS.Boot.DeltaSec
    Value: 3996

    Key  : Timeline.Process.Start.DeltaSec
    Value: 63

PROCESSES_ANALYSIS: 1

SERVICE_ANALYSIS: 1

STACKHASH_ANALYSIS: 1

TIMELINE_ANALYSIS: 1

Timeline: !analyze.Start
    Name: <blank>
    Time: 2019-06-04T19:04:14.834Z
    Diff: 94834 mSec

Timeline: Dump.Current
    Name: <blank>
    Time: 2019-06-04T19:02:40.0Z
    Diff: 0 mSec

Timeline: Process.Start
    Name: <blank>
    Time: 2019-06-04T19:01:37.0Z
    Diff: 63000 mSec

Timeline: OS.Boot
    Name: <blank>
    Time: 2019-06-04T17:56:04.0Z
    Diff: 3996000 mSec

DUMP_CLASS: 2

DUMP_QUALIFIER: 400

MODLIST_WITH_TSCHKSUM_HASH:  7d23a612ccb38087097364164db9a9af62e2fdd0

MODLIST_SHA1_HASH:  cebf23aa8222116acc7c3a2db099c01da3e3d563

NTGLOBALFLAG:  0

PROCESS_BAM_CURRENT_THROTTLED: 0

PROCESS_BAM_PREVIOUS_THROTTLED: 0

APPLICATION_VERIFIER_FLAGS:  0

[... 196 lines cut ...]

OSNAME:  Windows 10

OSEDITION:  Windows 10 WinNt SingleUserTS

USER_LCID:  0

OSBUILD_TIMESTAMP:  unknown_date

BUILDDATESTAMP_STR:  190318-1202

BUILDLAB_STR:  19h1_release

BUILDOSVER_STR:  10.0.18362.1.amd64fre.19h1_release.190318-1202

ANALYSIS_SESSION_ELAPSED_TIME:  3b27e

ANALYSIS_SOURCE:  UM

FAILURE_ID_HASH_STRING:  um:breakpoint_80000003_win32u.dll!ntusergethdevname

FAILURE_ID_HASH:  {e0b4ec90-6d57-8752-b67a-0e26c0df3fb2}

Followup:     MachineOwner
```

GitSoks, 2019-06-25:
I have the same Issue....

### Version
PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Windows 10 Professional 64 bit

### 3D printer brand / version + firmware version (if known)
Genuine Prusa i3 MK2.5, firmware 3.7.1

### Behavior
* **Problem**: Every time the PrusaSlicer window tries to open in any other Monitor then the main one (I have 3 monitors connected) it just gets stuck while loading the Interface (= a white Window opens without the Interface and the Mouse pointer indicates loading)

The only solution if this happens was for me to delete all PrusaSlicer config files in the "AppData" Windows folder so it opens on the main monitor again (resets the window Position do default = main monitor)

* **Workaround**:I have to always make sure I keep the PrusaSlicer window on the main Monitor and close it while it is on the main one. If I do that the next time I open PrusaSlicer, It has kept the Position and opens on the  #main Monitor again and everything works fine.

bubnikv, 2019-08-01:
May it be the same issue as https://github.com/prusa3d/PrusaSlicer/issues/2522 
?

gnydick, 2019-09-18:
Still happening, just downloaded the latest 2.1

bubnikv, 2019-09-19:
@gnydick Does it happen if you kill the Nahimic driver?

https://forum.prusaprinters.org/forum/prusaslicer/pursaslicer-2-0-0-not-responding/paged/2/#post-163791

Streeter1981, 2020-07-11:
I was having the same problem where PrusaSlicer 2.2.0 would not load on my main monitor, rather would just show a white screen. 

I just verified that the Nahimic service (audio service from ASUS motherboard) was indeed what was causing the problem. 

By running services.msc and stopping the Nahimic service, PrusaSlicer loaded just fine. When I re-enable it, it will not load. I'm going to switch this service to "Disabled" unless I find it's needed for something, but I doubt that it is.

gnydick, 2020-07-12:
Good to know. Still think PS shouldn't conflict with other software. The
question is, is PS acting too greedily or is the Nahimic service?

neophyl, 2020-07-13:
The nahimic drivers have been causing issues not just with Slicer but with other software too.  A simple google search reveals lots.  They dont like some msi drivers, nvidia driver etc etc.

goul, 2020-10-17:
I have a 3 screen win 10  system and after closing PrusaSlicer 2.2.0 down on a non primary screen the app would go to "Not responding" on every attempt to startup. Quick fix for me to get it back to the main screen was to hunt down PrusaSlicer.ini in \Users\<username>\AppData\Roaming\PrusaSlicer and comment out the line window_mainframe 

Then with a restart the app was on the main screen.

DommDynamite, 2021-06-03:
I cannot seem to open the slicer unless I only have one monitor plugged in. Sends me into a rage every time. What is with this? Why can't you guys get this handled? This is my second PC with all new hardware and I still have to unplug monitors to start your program??? WHY???

gnydick, 2021-06-04:
There's a common driver, nahimic, that installs a service. Disable the
service and you're good to go.

bubnikv, 2021-07-30:
Warning about the Nahimic DLL being injected into PrusaSlicer will be rolled out with PrusaSlicer 2.4.0-alpha1. It is not a "fix" for the nahimic issue, but at least the user will be informed.

closed 2021-07-30 by bubnikv