prusa3d/PrusaSlicer #2143: Line colors in preview mode

guestisp, 2019-04-19:
Is this a simple optical effect or something that will be printed differently ? Because I see some strangeness in my print more or less where the preview show the layer lines with a different color.

Looking carefully, the gradient used in the preview seems to be reversed in some areas.
Any particular reason ?

[image]

bubnikv, 2019-05-04:
It is the result of the particular discretisation of the extrusions, and how they share normal vectors across successive linear segments to make the visualisation smoother and to conserve system memory. It is just a visualization artifact, but we shall fix it as it leads to quite some error reports, where people believe that they will see the same artifact on the print.

guestisp, 2019-05-04:
Ok. So, should I close this issue or leave it open ?

bubnikv, 2019-05-04:
I think we can close it as there are more issues like this.

closed 2019-05-04 by bubnikv

guestisp, 2019-05-04:
Ok but please reply to the question if possible.

Anyway, at which others issues are you referring to?