prusa3d/PrusaSlicer #2144: Bad estimated printing time

guestisp, 2019-04-19:
### Version
1.42.0-beta2

When slicing using speeds in %, the estimated printing time is wrong. 
Probably because speeds are expressed in % and resultis in decimal values

in exampe: 

infill 60 mm/s
solid infill 85% (51mm/s)
top solid infill 30% (15.3mm/s)

In this particular case, setting top solid infill to an integer , like 15, fix the issue.
I think slic3r should automatically round the speeds to the nearest integer value.

[image]

bubnikv, 2019-04-23:
please attach a 3mf to reproduce

guestisp, 2019-04-23:
In this moment i'm unable to reproduce. I'll attach a 3mf as soon as I hit the issue again. Very strange, as I was able to create the screenshot so the issue is real.

guestisp, 2019-05-14:
Related to https://github.com/prusa3d/PrusaSlicer/issues/1380
Closing

closed 2019-05-14 by guestisp