prusa3d/PrusaSlicer #2145: New versions erase all settings

swissfreek, 2019-04-20:
### Version
_1.42.0-beta2+_

### Operating system type + version
_MacOS Mojave 10.14.4_

### 3D printer brand / version + firmware version (if known)
_Prusa i3 MK2.5S_

### Behavior
*All new version installs of Slic3R result in wiped settings and profiles. Guys, for the love of all that is holy, installing a new version of the software should not nuke all my installed profiles and settings. Either build a proper installer or make it work like almost every other Mac software, where it keeps the settings in the Library or some such. This is a horrendous oversight. So many hours of work and frustration and profile testing lost every time you release an update and I forget to specifically save my config files because there's no automatic backup, no profile saving, no configuration snapshots, and no persistent preferences files. Pain in the ass!

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

x0rtrunks, 2019-04-20:
They aren't nuked, they're just in a seperate folder. This was done intentionally starting with the alphas so that people could test the new builds without interfering with their stable install.

From the 1.42 alpha 1 release notes:
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha1

>To let you enjoy the alpha without worries, the alpha builds now save their profiles into Slic3rPE-alpha directory, so you may use the alpha side by side with the current release without ruining your production configuration.

swissfreek, 2019-04-20:
I stand corrected. My apologies. Had a moment of frustration and shock when I updated to the beta. Disregard.

closed 2019-04-21 by lukasmatena