prusa3d/PrusaSlicer #2146: Corrugated infill

espr14, 2019-04-20:
1.42.0 b2 win64

Rectilinear support infill uses straight lines which tend to bend in large areas. If you switch to wavy line it will make walls much more sturdy (see the image). It will preserve ease of support removal.

This approach can be used for all other infills. But it does not work for alternating ones (rectilinear, lines, 3D honeycomb) because printer cannot print unsupported curved line. Also it makes no sense to apply curves to already curved infills (Gyroid, Arch. chords).

[image]

bubnikv, 2019-05-04:
I don't understand the proposal, but let's others to chime in (or not).

guestisp, 2019-05-04:
> I don't understand the proposal

neither do I :)

espr14, 2019-05-04:
[image]

wavexx, 2019-06-17:
A wavy line will support more load perpendicular to the direction of the infill, yes, but it does so at the expense of inferior rigidity in the others.

You should simply use triangle or grid with a denser pattern if you want to increase the vertical loading strength. The infill intersections will protect against layer shearing just as efficiently.

Use cubic or gyriod if you want uniform loading characteristics across all axes.

This would only make sense if you were explicitly after the different compression response on the horizontal plane...

espr14, 2019-06-18:
This is meant for support infill in the first row. So there's no need for horizontal strength. But you need curved plane for easy removal which is not possible with intersecting infill. Also there are no gyroid or cubic etc. infills available for support infill.

wavexx, 2019-06-18:
I didn't get this was meant for supports. I actually do see a benefit with this.
As you say, long and tall support walls tend to sag quickly using "rectilinear".
"Rectilinear grid" takes longer and is generally very poor.

A simple sinus pattern will be a lot faster than honeycomb, and doesn't even need to be full-width.
Looks like something @supermerill could have some ideas on.

tuxmobil, 2019-09-22:
I like the idea. I just had some tall supports which failed (PrusaSlicer 2.1) and was searching for a way to solve this in the future.

[image]

The supports generated consist of a single straight line and didn't make it to height they were needed. They fell apart well before and I think the wavy pattern would have helped to make them more stable.

wavexx, 2019-09-22:
The same goes for high skirts of large rectangular shapes.
The large, single-wall edges start to collapse under their own weight at some point.

closed 2026-09-29 by prusaslicer-bot[bot]