prusa3d/PrusaSlicer #2147: Solid infill pattern setting

espr14, 2019-04-20:
1.42.0 b2 win64

While you can change top and bottom solid infill pattern, there is no way how to change internal solid infill. It's always rectilinear. Can you add a setting for this?

If you use Archimedean chord for circular object it is kind a vase mode for circular layer. This brings higher print speed and stronger layer. Other solid infill layers can be also Archimedean chord. The only change would be initial rotation of 180° per layer. This would lead to placing thread between two threads below which could increase strength.

bubnikv, 2019-05-04:
I believe this is a duplicate of https://github.com/prusa3d/Slic3r/issues/28 as the proposal in https://github.com/prusa3d/Slic3r/issues/28 is much better than what has been proposed here.

I am leaving this proposal open with the "very unusual idea" tag attached for others to chime in, but I propose to implement https://github.com/prusa3d/Slic3r/issues/28 instead of making the internal solid infill configurable.

espr14, 2019-05-04:
#28 is about perimeters. This is about solid infills.

bubnikv, 2019-05-04:
no, they are both about solid infill.

SharifZaman, 2019-08-25:
I'm with espr14 - I have a part with a color indicating it is a Solid infill.  It is the rectilinear pattern.  I would like to change it to the same as the top and bottom infills that I'm using, Octagram Spiral.  

[image]

0ctane, 2019-11-17:
I'm using PrusaSlicer-2.1.0+win64-201909160915.  I was trying to create a hilbert curve solid infill, and I came across (what I think is) the same issue.  While all (Fill Pattern, Top fill pattern, and Bottom fill pattern) my Infill Print Settings were configured for Hilbert Curve, I saw that there were rectilinear layers added.
[image]
The attached image is an example of a stepped structure.  My real intent is to have 100% infill as hilbert curve, which does slice and not have bridging layers, but it still produces a rectilinear layer right before the top layer.

0ctane, 2020-09-02:
I came across this again when using PrusaSlicer on a different model.  I double checked my horizontal shells, and I had 5 Solid layers set for the bottom.  However, even the second layer was rectilinear although I had all patterns set to Hilbert Curve.

Any movement on this issue?

metaColin, 2021-04-25:
*bump*

I am also very interested in this feature. I am experimenting with translucent filament and how it conducts light.

I very much need to be able to control the 'solid fill pattern' alongside top and bottom.

larsinka, 2021-05-25:
I'd love to see this added as well. It would be a lot quicker in my batch production.

[image]

imnotme, 2021-05-31:
Same, very interested to see this implemented. Have to use modifiers, and it's very time consuming

ty

P1SV, 2021-08-25:
Same with me,
I am trying to do the valve and I really need concentric infill for its properties. Is there around at least some dirtyhack?

[image]
[image]

brettonidas, 2023-10-02:
4 years later, this would still be helpful. I'm printing circular items, and concentric internal and solid infill would be amazing.

AntumArk, 2023-10-16:
> Same with me, I am trying to do the valve and I really need concentric infill for its properties. Is there around at least some dirtyhack?
> 
> [image] [image]

Set perimeters to 100 and it should work. Dirty hack :D
[image]

pilot51, 2024-08-06:
I've been printing PHA which really likes to warp. One way to reduce warping is to avoid infill with straight lines. For many objects, I want the visible top and bottom layers to be straight and to prevent warping I want the rest of the infill to be a pattern other than rectilinear or monotonic. This feature would help.

Behem07h, 2024-12-22:
i'm also interested in this feature, I'm trying to get clear filament to print clear and being able to control the solid infill pattern will be very useful for minimising air bubbles in odd shaped parts

toomanydev, 2025-01-13:
I would very much like this feature as well. 
Using Hilbert Curve on the solid infill may help with warping of the bottom layer, which would be good for printing PHA. PHA seems to have a unique feature where it warps more with high ambient temperature and less with low ambient temperature, which means it can't be solved with an enclosure like regular temperature-resistant materials. Currently, the second, third, etc layers will still have that tendency to pull inward even with setting all available infills to Hilbert.

perpleksed, 2025-01-19:
Why is this tagged as "very unusual idea"? 😮

I initially though it was a **bug**, when I changed bottom and top infill patterns to monotonic lines, but it affected only the first and the last one layers. Decided to google it first and stumbled upon this abandoned issue. Obviously I would prefer all bottom and top layers printed in the selected pattern, it just makes sense.

 OrcaSlicer at least has an option to change "Internal Solid Infill Pattern" separately

pilot51, 2026-02-25:
@Jealow Removing the top and bottom internal solid infill is not really a solution. I don't think anyone wants to sacrifice shell integrity just to avoid a forced pattern. It is only a solution if fill density is 100% and we want the top and bottom internal solid infill to be the same pattern as the rest of the internal infill.

Jealow, 2026-02-25:
@pilot51 Yes I meant it for 100% infill. I used it for printing long and thin pieces. It's kind of shocking that we can't choose solid infill pattern as easily as infill pattern.

perpleksed, 2026-09-02:
This issue persists in v3.0.0-alpha.

<details><summary>Details</summary>
<p>

[image]

[image]

</p>
</details> 

Currently, there is no clean way to set top and bottom infill patterns. The settings apply only to the outermost layers, rather than the entire top / bottom infill.