prusa3d/PrusaSlicer #2148: Make Concentric infill continuous

espr14, 2019-04-20:
1.42.0 b1 win64

Concentric patter is great for rectangular shapes like a Archimedean chord for circular ones. But it is a set of independent rectangles while it is easy to make them continuous. That would bring better bed adhesion and faster print because of less retractions and turns.

[image]

supermerill, 2019-04-23:
could you draw an example of the path to create? I can't visualize a better pattern than the current one.

espr14, 2019-04-23:
[image]

bubnikv, 2019-05-04:
Finding such a curve for a general polygon is non-trivial. I have seen a paper I believe in Siggraph, that described such approach. Their results were not convincing to me, as they were leaving a lot of gaps, and it was very computationally intensive.

espr14, 2019-05-04:
Concentric already leaves lots of gaps. The connection points are at the extreme angles. Computational intensity like aligned seams.

bubnikv, 2019-09-05:
> The connection points are at the extreme angles. Computational intensity like aligned seams.

Please provide a pull request then.

espr14, 2019-09-05:
Sorry, I don't have spare time to code.

closed 2026-09-29 by prusaslicer-bot[bot]