prusa3d/PrusaSlicer #2149: Center infills to local center

espr14, 2019-04-20:
1.42.0 b1 win64

While concentric infill computes center from the local area, Archimedean chord and Octagram spiral is not. Could it put the center of the spiral to the local center of object? Following shapes are from single object.

Concentric:
[image]
Archimedean chord:
[image]

bubnikv, 2019-05-04:
I am marking such an issue with the "very unusual idea" tag. While such an idea may be useful to somebody once upon a century, adding such rarely used features into software will quickly lead to a usability and maintenance nightmare. I will leave such issues open for some time. If nobody chimes in, I will close them.

bobmonkey07, 2019-08-17:
[image]
I had noticed this too. It's not critical by any means, but it would be nice on occasion.

KyleMaas, 2020-01-19:
I use Archimedian Chords quite a bit, and this would be really useful for me.  I've had to change the shapes of objects (add junk to one side, for example) to push the infill center where I want it.

James-G4CLF, 2020-05-15:
In general it would be helpful to define infill positions more precisely.

bubnikv, 2020-12-09:
> In general it would be helpful to define infill positions more precisely.

Picking the center point with mouse would be nice, however there is no upper limit to the software complexity and the day has 24 hours only.

MonoTovarisj, 2020-12-09:
> > In general it would be helpful to define infill positions more precisely.
> 
> Picking the center point with mouse would be nice, however there is no upper limit to the software complexity and the day has 24 hours only.

Well, I think it would be better to be able to select a circular face or edge to define the center, and in case of a square the center of 4 edges. I know that might not make it simpler but I think it will be hard to define it precisely with a mouseclick, even though it would be a big improvement.

pmcsr, 2021-01-07:
Has anyone found a solution to this? 
[image]
I would love to be able to center the octagram spiral for the first layer pattern. In this instance the first layer is actually the top of the finished part, so the irregularity stands out.

James-G4CLF, 2026-09-18:
Maybe too complex, but it would be useful to move centres or other reference points in these layers. Don't kill it - simplify and do it.