prusa3d/PrusaSlicer #2150: Doesn't build with Boost 1.70

lovesegfault, 2019-04-21:
I'm trying to build this project in order to package it for Gentoo, but the build is failing, and I suspect it is due to my system having the recently released Boost 1.70. Here's the compilation error:
```
/home/bemeurer/src/Slic3r/src/slic3r/Utils/Serial.cpp: In member function ‘bool Slic3r::Utils::Serial::read_line(unsigned int, std::__cxx11::string&, boost::system::error_code&)’:
/home/bemeurer/src/Slic3r/src/slic3r/Utils/Serial.cpp:387:21: error: ‘get_io_service’ was not declared in this scope
  auto &io_service = get_io_service();
                     ^~~~~~~~~~~~~~
/home/bemeurer/src/Slic3r/src/slic3r/Utils/Serial.cpp:387:21: note: suggested alternative: ‘io_service’
  auto &io_service = get_io_service();
                     ^~~~~~~~~~~~~~
                     io_service
```

vojtechkral, 2019-04-23:
Thanks for the report, but how can we fix this? Boost doc suggest using `get_executor()`, but that's quite new and not present in not-that-old Boost versions, such as the ones in [Debian Jessie and Stretch](https://packages.debian.org/search?searchon=names&keywords=libboost) and also Ubuntu LTS I think.

If you replace `get_io_service()` with `get_io_context()`, does it build against Boost 1.70?

Johnnynator, 2019-05-23:
> If you replace `get_io_service()` with `get_io_context()`, does it build against Boost 1.70?

They are both removed in boost 1.70 https://github.com/boostorg/asio/commit/a72fbb0b867fd8fb37a233d893c3f5e063e72ea6 so it won't help

vojtechkral, 2019-05-23:
FWIW a lot of the functionality in that file is not used anymore, we should probably just remove it. That would solve the problem. I'll put it on my todo list...

obfuscated, 2019-07-31:
@vojtechkral  Any progress with this issue?

vojtechkral, 2019-08-02:
@obfuscated Sort of. I wanted to do a proper reimplementation, but there's no time right now, so I just commented out the offending code. It's ugly, but at least it should build now with boost 1.70.

commit bfb135bc (2019-08-02): Comment out stale implementation in Serial, fix #2150
closed 2019-08-02 by vojtechkral