prusa3d/PrusaSlicer #2151: Number of perimeters for solid infill layers

espr14, 2019-04-21:
1.42.0 b2 win64

Perimeters are kind of solid infill. Perimeters copy shape better but infill looks better in the area. So it would be good to allow to decrease number of perimeters for areas with solid infill. Even no perimeter makes sense.

[image]
[image]
[image]
[image]

supermerill, 2019-04-23:
> Perimeters copy shape better but infill looks better in the area.

gap-fill is needed between perimeters, because it's not a good way to fill an area. rectilinear is more efficient (but for some angles). Your last picture may be problematic, as the shell will look horrible.
If we want nice looking shell + efficient filling, the current solution of some perimeters+rectilinear is the best solution (imo).

espr14, 2019-04-23:
Last image is extreme. However, imagine this: I want all walls to be thick at least 10 perimeters. But with 10 perimeters the bottom and top layers will look weird. Since the solid infill make the layer strong enough, we don't need extra perimeters. One or two perimeters are needed to make the shape smooth and to have something the infill can catch on.

In case of circular object and Archimedean chord there's no need for perimeter at all.

supermerill, 2019-04-23:
>But with 10 perimeters the bottom and top layers will look weird. Since the solid infill make the layer strong enough, we don't need extra perimeters

I created a "only one perimeter on top surfaces", see https://github.com/prusa3d/Slic3r/pull/821
You can try it in my fork.
I don't know if it's really that useful for the bottom layer.

espr14, 2019-04-23:
Sometimes it is more useful for the bottom layer. For example parts for Prusa i3 MK2, MK3 are designed so the bottom layer is the visible one.

supermerill, 2019-04-23:
what i mean is that my bottom layer is pushed hard against the build plate and so i can't see any pattern on it..

espr14, 2019-04-23:
It's more about the material and manufacturer. In my case neither hard pressure nor 110 °C helps.

MagnusThome, 2023-02-13:
I really miss the option of one single perimeter on the top layer (like SuperSlicer has) 

As also mentioned here https://github.com/prusa3d/PrusaSlicer/issues/7986 with these pictures below really showing the benefit of this!

[image]

Jan-Soustruznik, 2025-04-11:
Implemented since 2.8.0-alpha5, closed as solved

closed 2025-04-11 by Jan-Soustruznik