prusa3d/PrusaSlicer #2152: 1.42.0beta2: support not generated properly (hollow to plate?)

Mirarkitty, 2019-04-21:
### Version
1.42.0beta2

### Operating system type + version
Linux, AppImage build

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
Support is not generated properly for large overhangs, not entirely sure what triggers it but a suspicion is closed hollows to build plate.

[image]

Turning off "_don't support bridges_" will make support happen, but introduce another issue, a **0.3mm (two layer!) gap between the support and overhang**, which makes the generated support useless:
[image]

[image]

#### Project File (.3MF) where problem occurs
[nosupport.3mf.gz](https://github.com/prusa3d/Slic3r/files/3101082/nosupport.3mf.gz)
[nosupport.stl.gz](https://github.com/prusa3d/Slic3r/files/3101087/nosupport.stl.gz)

Mirarkitty, 2019-04-21:
Cutting off a corner of the "frame" yields almost comical results:

[image]
[nosupportc.3mf.gz](https://github.com/prusa3d/Slic3r/files/3101094/nosupportc.3mf.gz)

Mirarkitty, 2019-04-21:
Oh, also "support enforcer" modifier does not make any difference.

Mirarkitty, 2019-05-16:
Exist in PrusaSlicer-rc as well: 
[image]

bubnikv, 2021-02-22:
You have "don't support bridges" enabled. If you disable the option, then the supports are being generated.

closed 2021-02-22 by bubnikv