prusa3d/PrusaSlicer #2153: Infill interface

espr14, 2019-04-21:
1.42.0 b2 win64

Direct transition from sparse infill to solid infill is always fixed. No matter what infill type and density is selected. This causes overhangs which can be really long which leads to ugly top infills. Thus I propose infill interface. Something like support interface but only between sparse and solid infill. Since sparse infill knows where the threads are, it can build bridges instead of overhangs. This interface can have lower infill density (50 - 90 %) so the bridges can interleave and the same infill type can be used (most of types cannot reach 100 %).

[image]
[image]
[image]

supermerill, 2019-04-23:
(sort of) duplicate of https://github.com/prusa3d/Slic3r/issues/569

also see https://github.com/slic3r/Slic3r/issues/4778

espr14, 2019-04-23:
I propose not only changing the shape but also could be of different infill type and density. But it's the same basis.

closed 2019-04-23 by espr14