prusa3d/PrusaSlicer #2155: Sheath does not show up

espr14, 2019-04-21:
1.42.0 b2 win64

If I turn on support sheath nothing changes.

FidelCapo, 2019-04-23:
I can not reproduce your issue. I think it works correctly.

support sheath enabled
[image]

support sheath disabled
[image]

espr14, 2019-04-23:
I wasn't able to find what it does specifically. Still, it does not apply to the first layer. But let's solve this elsewhere.
#2101

closed 2019-04-23 by espr14